openhwfoundation/core-v-verif #8: Platform dependent casting errors

MikeOpenHWGroup, 2019-12-16:
This issue affects Verilator simulations only.

Line 76 of cv32/tb/core/tb_top_verilator.cpp yields the following compile error under Ubuntu 18.04:
`error: cannot convert ‘svLogicVecVal* {aka t_vpi_vecval*}’ to ‘const svBitVecVal* {aka const unsigned int*}’ for argument ‘1’ to ‘int read_byte(const svBitVecVal*)’
             uint32_t val = read_byte(&addr);`

This error does not exist under Debian.  I've added a "solution" to line 77 (commented-out) that works in Ubuntu, but fails under Debian.

This issue exists to ensure I find a real solution that works for both Debian and Ubuntu.

Smattacus, 2020-05-01:
This issue happens when verilator is installed from the Ubuntu 18.04 repositories using apt. Building verilator from source ([section 2.3 here](https://www.veripool.org/projects/verilator/wiki/Installing)) fixes it. I will submit a pull request to add a note to the documentation.

MikeOpenHWGroup, 2020-05-01:
Bingo!   That resolves the issue, thanks @Smattacus.   I built `Verilator 4.032 2020-04-04 rev v4.032` from source and the cast issue cleared up.

closed 2020-05-01 by MikeOpenHWGroup