openhwfoundation/core-v-verif #11: Hello World: Elaboration errors with older Cadence Xcelium

GTumbush, 2020-01-03:
hello_world test requires Xcelium version 19.09 or higher.

The error report below is from Xcelium 19.03-s002

A suggested fix is to re-code the structure as a class and pass objects through the mailbox. 

**Working in:** core-v-verif/cv32/tests/core
**Executing:** make xrun-hello_world
**xmelab errors with:**
mailbox #(stall_mem_t) core_reqs          = new (4);
                    |
xmelab: *E,MBXNYI (/wrk/gtumbush/mcu/ver/core-v-verif/cv32/tb/tb_riscv/riscv_random_stall.sv,87|20): 'unpacked structure riscv_random_stall::stall_mem_t (stall_mem_t, file /wrk/gtumbush/mcu/ver/core-v-verif/cv32/tb/tb_riscv/riscv_random_stall.sv, 77)' is not a supported type for mailboxes.
 
**Help on this error:**
nchelp xmelab MBXNYI
xmhelp: 19.03-s002: (c) Copyright 1995-2019 Cadence Design Systems, Inc.
xmelab/MBXNYI =
                Parameterized mailboxes currently support class and all the predefined
                data types such as bit, logic, int and their arrays as it's parameter
                type. Rest are unsupported yet.

**Referring to the Cadence release notes for Xcelium 19.09:**
The page contains the CCRs fixed in this release. 
XCELIUM Main Version 19.09
December, 2019
997972      CORE_ELAB       Support for unpacked structure as a datatype for mailbox

MikeOpenHWGroup, 2020-01-06:
Hi Greg, thanks for raising this issue.

I'm inclined to mark the issue as "wont fix" since we know that 19.09 works and your team has access to that version.   Another member company (BTA) is using Xcelium 19.09.s003 and they have confirmed that the testbench compiles as-is in their environment.

I certainly agree that unpacked struct typedefs for SV mailboxes are a bit of an esoteric feature of the language, but its not a new feature, so I do not see any benefit going through the effort to make the testbench backward compatible with an older version of Xcelium.

I've labeled this issue as **wontfix** and assigned it back to you.  If you agree with this, please close the issue.  Otherwise, please remove the label, re-assign it back to me and provide a rationale for making the existing testbench backward compatible with Xcelium 19.03.

GTumbush, 2020-01-06:
Agree to not fix the code but should still be documented as to the minimum Xcelium or version of Xcelium testing was done with.

closed 2020-01-06 by GTumbush

MikeOpenHWGroup, 2020-01-07:
Thanks Greg.  I've updated the Makefile at cv32/tests/core.