openhwfoundation/core-v-verif #25: dsim: symbol lookup error: when compiling UVM

MikeOpenHWGroup, 2020-01-27:
I've committed a bare-bones UVM testbench and testharness at cv32/tb/uvmt_cv32 (no env class yet).  Compilation proceeds as you'd expect but terminate with a `symbol lookup error`.

### Expected behavior
In this "test" all we are doing is compiling the code, so we are looking for the absence of compile errors.

### To Reproduce
Steps to reproduce the behavior:
1. Create a branch from https://github.com/openhwgroup/core-v-verif
2. cd cv32/tb/uvmt_cv32
3. make (default rule is the one you want)

### Terminal dump:
```
mike@virtualserver01:/data/mike/GitHubRepos/openhwgroup/core-v-verif/master/cv32/tb/uvmt_cv32$ make
mkdir -p /data/mike/GitHubRepos/openhwgroup/core-v-verif/master/cv32/tb/uvmt_cv32/results
mkdir -p /data/mike/GitHubRepos/openhwgroup/core-v-verif/master/cv32/tb/uvmt_cv32/results/dsim_work
dsim \
	-timescale 1ns/1ps \
	+incdir+/tools/Metrics/dsim/20191112.8.0/uvm-1.2/src /tools/Metrics/dsim/20191112.8.0/uvm-1.2/src/uvm_pkg.sv \
	-f /data/mike/GitHubRepos/openhwgroup/core-v-verif/master/cv32/tb/uvmt_cv32/uvmt_cv32.flist
Copyright (C) 2017-2020 Metrics Technologies Inc. All rights reserved.
This program is licensed for the exclusive use of imperas,
and may not be used otherwise without written permission from Metrics Technologies Inc.
Analyzing...
Elaborating...
  Top-level modules:
    $unit
    uvmt_cv32_tb
  Found 5 unique specialization(s) of 5 design element(s).
Optimizing...
Building models...
Simulation time precision is 1ps.
  [1/9] module $root: 1224 functions, 2772 basic blocks
  [2/9] module $pkgs: 1 functions, 2 basic blocks
  [3/9] CU scope $unit: 1 functions, 2 basic blocks
  [4/9] module uvmt_cv32_tb: 5 functions, 21 basic blocks
  [5/9] package std: 31 functions, 80 basic blocks
  [6/9] package uvm_pkg: 4 submodules, 3612 functions, 26796 basic blocks
  [7/9] package uvmt_cv32_pkg: 1 functions, 2 basic blocks
  [8/9] interface uvmt_cv32_clk_gen_if: 5 functions, 21 basic blocks
  [9/9] module uvmt_cv32_dut_wrap: 1 functions, 2 basic blocks
Linking image.so...
=S:Begin run-time elaboration and static initialization...
dsim: symbol lookup error: dsim_work/image.so: undefined symbol: uvm_dpi_get_next_arg_c
Makefile:79: recipe for target 'dsim-comp' failed
make: *** [dsim-comp] Error 127
mike@virtualserver01:/data/mike/GitHubRepos/openhwgroup/core-v-verif/master/cv32/tb/uvmt_cv32$ 
```

aimeepsutton, 2020-01-28:
It looks like your code compiled successfully. The error occurs because DSim has proceeded to the simulation run phase, but is missing the UVM DPI library (which is loaded using the run-time argument `-sv_lib $UVM_HOME/src/dpi/libuvm_dpi.so`).

If you want DSim to stop after compilation is complete, use the `-genimage` option and provide a name for the compiled image.

Let me know if you have any further questions!

MikeOpenHWGroup, 2020-01-28:
Thanks Aimee. That worked!  Closing the issue.

closed 2020-01-28 by MikeOpenHWGroup