openhwfoundation/core-v-verif #33: Implement CLK&RST Agent and Virtual Sequence for cv32 env

MikeOpenHWGroup, 2020-02-06:
Currently the clocks and resets for uvme_cv32 are contained entirely within `interface uvmt_cv32_clk_gen_if ` (in file cv32/tb/uvmt_cv32/uvmt_cv32_tb_ifs.sv).  The procedural code for this should be moved to an agent and controlled in a vseq as part of the base_test.

We want randomization of:
- clk period
- initial value of clk signal ( 0, 1, X)
- initial value of rstn (0, 1, X)
- if the initial value of rstn is 1 or X, a random de-assertion duration
- random assert duration for rstn

We also want to preserve the start() and period() methods to control the clock (these can be implemented in the SV interface.

As we discussed, having a full-blown UVM Agent, and Virtual Sequences for this is somewhat overkill, but this will be used as an example for other team members, so I'd like to get in this quickly.   Lots of comments please!

datum-dpoulin, 2020-03-02:
I have the agent skeleton generated. Started work on implementation.

datum-dpoulin, 2020-03-04:
Do we want the reset de-assert to be specified in clock cycles or time units?

MikeOpenHWGroup, 2020-03-23:
Closing, see #82 

Thanks David.

closed 2020-03-23 by MikeOpenHWGroup