openhwfoundation/core-v-verif #37: Move RI5CY RTL to CORE-V-CORES

MikeOpenHWGroup, 2020-02-10:
### Task Outcome
Move (not clone) the RTL model currently in the PULP Platform project at https://github.com/pulp-platform/riscv to the OpenHW Group project 

### Location Information
The new location should be a CV32E40P directory under https://github.com/openhwgroup/core-v-cores.  

### Completion Criteria
Code has been moved and compiles as before in its new location.

### Additional context
Some modifications to the structure of the RTL code will be necessary.  See https://github.com/openhwgroup/core-v-verif/tree/master/rtl for an example.

davideschiavone, 2020-02-11:
Hi @MikeOpenHWGroup ,

> The new location should be a CV32E40P directory under https://github.com/openhwgroup/core-v-cores.

The new location will be https://github.com/openhwgroup/cv32e40p repository and not a folder inside that repo. Otherwise we end up having many cores to the same repository. This makes maintaining the repo unfeasible.

davideschiavone, 2020-02-12:
The core has been transferred.

You can close the issue

MikeOpenHWGroup, 2020-02-12:
This is a big deal for OpenHW.  Thanks to you and the whole PULP team!

closed 2020-02-12 by MikeOpenHWGroup