openhwfoundation/core-v-verif #40: Add sanity and regression rules for simulation Makefiles

MikeOpenHWGroup, 2020-02-10:
### Task Outcome
The simulation Makefiles should support the following rules:
1. `make dsim-sanity`
2. `make dsim-regression`
3. `make xrun-sanity`
4. (optional) `make xrun-regression`

### Location Information
Simulations using the `core` testbench are run from `cv32/sim/core`.
Simulations using the `UVM` testbench are run from `cv32/sim/uvmt_cv32`.

### Completion Criteria
Rules should run successfully for both `core` and `uvmt_cv32` Makefiles.

MikeOpenHWGroup, 2020-02-18:
Hi Aimee.  I am assigning this to you for comment.  In the `Task Outcome` there are proposed rules for both `sanity` and `regression` testing.  Does that make sense given that the Metrics CI rules are where the regression is setup?   In that case, maybe all we want is a sanity rule.

aimeepsutton, 2020-02-20:
Hi Mike, I agree that all we want is a sanity rule, provided that sanity just runs a single test and not a sanity regression. Metrics defines regressions in a separate file, and it's fair to assume that member organizations will have their own regression infrastructure (e.g. scripts, job submission  system). It would not be reasonable to include support for all of these in the Makefile.

What I suggest we do is come up with a test list for each regression, and a format (CSV or other) for specifying the options/arguments for each test (e.g. random seed, etc). I can write a custom script to parse these test lists and output the format that Metrics requires. Member organizations can do the same: write a script that parses the testlist and formats it for consumption by their internal tools. This way everyone will be running the same regressions, but the actual mechanics of doing so can be different. 

Let me know if you have any additional thoughts or comments!

MikeOpenHWGroup, 2020-03-02:
This is largely complete.  The sanity tests are currently available on the master branch for the core testbench and the sanity tests are available for the uvmt_cv32 environment on the development branch.

As per the discussion with Aimee, we will forego adding regression targets in the Makefiles.

MikeOpenHWGroup, 2020-03-27:
Sanity tests available and known to be working for `dsim` and `xrun`.   Sanity for `vsim` (questa) is available, but not yet tested.

As of #80 the usage has been changed to: `make SIMULATOR=<sim> sanity`.  See the [README](https://github.com/openhwgroup/core-v-verif/blob/master/cv32/sim/uvmt_cv32/README.md#available-tests) for details.

closed 2020-03-27 by MikeOpenHWGroup