openhwfoundation/core-v-verif #46: Update RISC-V compliance testsuite

MikeOpenHWGroup, 2020-02-10:
### Task Outcome
Integrate the latest [RISC-V Foundation compliance testsuite](https://github.com/riscv/riscv-compliance) into the `uvmt_cv32` CV32E40P verification environment.  Stretch goal is to also integrated into the `core` testbench.

### Background information
The current version of the compliance testsuite was forked from the PULP-Platform RI5CY project in mid-December 2019.  We need to be running with the latest.

### Location Information
The current location for the compliance testsuite is `cv32/tests/core`.  It is recommended that the updated compliance tests be committed to `cv32/tests/compliance` and the existing tests deleted.

### Completion Criteria
1. Updated compliance tests installed as above.
2. README to capture Git hash of installed version.
3. Successful compile and regression of the compliance suite in the uvmt_cv32 environment.  (Not necessarily 100% passing).
4. Stretch goal: Successful compile and regression of the compliance suite in the core testbench with both **_dsim_** and **_Verilator_**.

Silabs-ArjanB, 2020-06-29:
Currently only rv32i seems to be included in the core-v-verif riscv-compliance tests. Can the following also be added please?

- rv32Zicsr 
- rv32Zifencei
- rv32im
- rv32imc

MikeOpenHWGroup, 2020-07-07:
Hi @Silabs-ArjanB, you are right, these tests are not working properly across all simulators.  This is a toolchain issue and I am working with the SW TG to resolve.

sjthales, 2020-07-07:
Hi @Silabs-ArjanB @MikeOpenHWGroup , actually the RISC-V compliance has been implemented with questa but it has been implemented to be used by all simulators.

To do so, in common.mk tests are run with this target  `%.run-cv32-new-compliance-test: %.${SIMULATOR}-run`

In vsim.mk, we added:
`%.vsim-run:`
	`@echo "sim: $*"`
`make SIMULATOR=vsim vsim-all VSIM_FIRMWARE=$*`

And you are right, there is no `%.xrun-run` target in xrun.mk. 
According to me, that should make sense to update xrun.mk in order to have a `%.xrun-run` target. I took a look on xrun.mk and I think this can easily be done.

Furthermore I also took a look on vcs.mk and dsim.mk I think that can be done quickly.

Regarding, the 'new' compliance, I have no opinion on the subject, maybe Mike could explain why he wants to keep the current compliance tests.

Regards,

MikeOpenHWGroup, 2020-07-07:
Hi @sjthales and @Silabs-ArjanB 

> Furthermore I also took a look on vcs.mk and dsim.mk I think that can be done quickly.

Agreed.  But it cannot be done cleanly.  :-)   I have a branch that has the new compliance tests working with a non-pulp toolchain.  I do not want to have two toolchains in the flow, so I have not integrated it in.  The SWTG is working to produce a "blessed" COREV toolchain that will support both PULP and standard instructions.   When this is deployed (and working) I'll integrate it all in.

> Regarding, the 'new' compliance, I have no opinion on the subject, maybe Mike could explain why he wants to keep the current compliance tests.

The "old" compliance tests will be retired when the new ones are on the master branch of core-v-verif and passing with dsim and xrun.

Yes, I know this is taking a long time - I am just as frustrated as anyone.

MikeOpenHWGroup, 2020-10-28:
This issue is now out of date (see the [README](https://github.com/openhwgroup/core-v-verif/tree/master/cv32/sim/uvmt_cv32#risc-v-compliance-test-suite) in the `cv32/sim/uvmt_cv32` directory of [core-v-verif]((https://github.com/openhwgroup/core-v-verif), so I am closing it.

closed 2020-10-28 by MikeOpenHWGroup