openhwfoundation/core-v-verif #53: Review the strategy for linking core-v-verif to the DUT

MikeOpenHWGroup, 2020-02-11:
### Task Outcome
An agreed method for linking the verification environments, supported in this GitHub project, to the Device(s) Under Test, supported in [cv32e40p](https://github.com/openhwgroup/cv32e40p) and other projects.

### Background information
Currently, the Makefiles have dependencies that clone the DUT repositories as necessary:
```
# CV32E40P RTL dependencies
$(CV32E40P_PKG):
	git clone https://github.com/openhwgroup/cv32e40p --recurse $(CV32E40P_PKG)
	git clone https://github.com/pulp-platform/fpnew --recurse $(FPNEW_PKG)
```
There are a number of issues with this:
1. It is not obvious to the maintainers of these other GitHub repositories that core-v-verif is dependant upon them.  Changes to either repo can break core-v-verif.
2. It can make it difficult to reproduce testcase behavior because the external clones can change.
3. We might simulate and synthesize different code.

There are multiple solutions to all of these issues.  We need to select the best one.

### Completion Criteria
Consensus on a solution for the above issues (and any others that may be posed).

### Additional context
If this issue is assigned to you, please comments and re-assign back to Mike Thompson.

MikeOpenHWGroup, 2020-03-30:
I have implemented the following in [Common.mk](https://github.com/openhwgroup/core-v-verif/blob/master/cv32/sim/Common.mk), the common makefile for core-v-verif simulations.

The Makefile contains the following variables and uses these to create a `git clone` command that clones both the [cv32e40p](https://github.com/openhwgroup/cv32e40p) and [fpnew](https://github.com/pulp-platform/fpnew) RTL sources.
```
CV32E40P_REPO   ?= https://github.com/openhwgroup/cv32e40p
CV32E40P_BRANCH ?= master
CV32E40P_HASH   ?= 9cdf35c8c460a933496b84c5b51f88652981fd5d

FPNEW_REPO      ?= https://github.com/pulp-platform/fpnew
FPNEW_BRANCH    ?= master
FPNEW_HASH      ?= da5fd4f0140c45f652c4a82a193f017484e3c72e
```
To compile the RTL, the Makefile uses the [manifest](https://github.com/openhwgroup/cv32e40p/blob/master/cv32e40p_manifest.flist) from the cv32e40p repository.

This strategy addresses each of the above issues:
- **Issue 1**: Changes to the design repos will not break core-v-verif because the Makefile will always clone from a known-stable version of the RTL.
- **Issue 2**: The REPO, BRANCH and HASH information will ease reproducibility issues.
- **Issue 3**: Use the same manifest on the same repo/branch/hash.

Hi @davideschiavone, do you have any comments on this strategy?

davideschiavone, 2020-04-02:
The strategy looks ok for me.
Another thing we should try to do is to decouple the FPU from CV32.
This should be possible by:

1. Move the RVF decoder outside the CV32.
This implies extending the APU interface and to limit the current CV32 decoder to only recognize whether there is an RVF (any RVF) instruction or not and to read the Register File. The Operands and instruction binary should be sent outisde the core through the APU interface where the FPU does the rest of the job (i.e. decoding and executing the instruction). The result is then sent back to the APU interface to be stored in the CV32 core.
In this way the CV32 repo does not depend on the FPU anymore.
I ask @stmach to please confirm it (as owner of the FPU repository).

stmach, 2020-04-02:
Yes this seems to be the best option to cleanly support also a non-FP version of the core without dangling dependencies to any FP-related packages, especially during compilation where you need the package even though you're parametrizing to not use the FPU later.
In general, formalizing this "APU interface" will greatly help to hook up anything to the core. You could probably parametrize the current one so that you can offload entire instructions fairly easily, without trying to create a fully generic coprocessor interface.

This RISC-V to FPU block (basically a decoder) should actually become part of the FPU repo to not drag `fpnew_pkg` dependencies out somewhere else.

MikeOpenHWGroup, 2020-11-02:
I am closing this issue now as the core-v-verif has settled on using Makefile variables create a git clone command that clones the cv32e40p RTL sources.  This strategy has been working well for several months.  Also as of openhwgroup/cv32e40p#560, the cv32e40p RTL no longer relies on an external fpnew package.

closed 2020-11-02 by MikeOpenHWGroup