openhwfoundation/core-v-verif #64: Questa/Vsim support for the uvmt_cv32 Makefile

MikeOpenHWGroup, 2020-02-18:
### Task Outcome
The core-v-verif repo has been updated such that the UVM verification environment can run the same "firmware" tests as the core testbench.  This has been tested with Metrics dsim.   It is important that these tests also can run with the Mentor SystemVerilog simulator.

### Background information
This task is related to issue [#52](https://github.com/openhwgroup/core-v-verif/issues/52)

### Location Information
The Makefile for the UVM environment is located at https://github.com/openhwgroup/core-v-verif/blob/master/cv32/sim/uvmt_cv32/Makefile.  It includes a common Makefile at https://github.com/openhwgroup/core-v-verif/blob/master/cv32/sim/Firmware.mk

### Completion Criteria
Ability to run the same [dsim tests](https://github.com/openhwgroup/core-v-verif/tree/master/cv32/sim/uvmt_cv32#running-the-envrionment-with-metrics-dsim) using Questa and/or Vsim.

sjthales, 2020-02-18:
@MikeOpenHWGroup I am working on the Makefile, where is the UVM_HOME defined?

MikeOpenHWGroup, 2020-02-18:
Good question!   Most EDA vendors will define UVM_HOME as part of their simulator configuration.  For example, the Metrics DSIM simulator defines it to be /tools/Metrics/dsim/20191112.8.0/uvm-1.2 because that is where the Metrics distribution of UVM is installed when you install DSIM.   I am aware that Cadence does something similar.  As a Questa user, you will need to either point to Mentor's UVM distribution or install your own version of UVM and use that.

I see that this is not mentioned in any of our READMEs, so I will fix that now.
Metrics is recommending that each AC use the UVM distribution from their company's tools, so for now we are following that advice.  What is your opinion?

agrasset, 2020-02-18:
We have the same opinion than Metrics here. We use the UVM distribution from Mentor's tools.

MikeOpenHWGroup, 2020-02-18:
Good.  Aimee will be pleased that you agree with her.  :wink:

I've updated the README at https://github.com/openhwgroup/core-v-verif/blob/master/cv32/sim/uvmt_cv32/README.md.  Thanks for pointing this out.

sjthales, 2020-02-18:
Do you use DPI in the UVM testbench?

MikeOpenHWGroup, 2020-02-18:
Not currently, but we will use the DPI to integrate the Imperas Instruction Set Simulator.  This work is starting now.

sjthales, 2020-02-21:
I am working on this issue, I encountered a problem with uvme_cv32_pkg and uvmt_cv32_pkg packages.
They are included each in the other, questa generates an error.
Is that normal that uvme_cv32_pkg is imported in uvmt_cv32_pkg and uvmt_cv32_pkg is imported in uvme_cv32_pkg?

MikeOpenHWGroup, 2020-02-21:
Whoops, that may be an error.  This _could_ have been my manually generated mistake, or it could be an issue with the templates we used to generate the ENV files.   Hi @datum-dpoulin, can you comment?

In the meantime, @sjthales, please resolve the issue on your branch and issue the pull-request when it works for you.  I'll have a look at your fix and see how Metrics **_dsim_** handles it.

MikeOpenHWGroup, 2020-02-22:
Hi @datum-dpoulin, here is what I _think_ @sjthales, is asking about.   This is where `uvme_cv32_pkg` is imported:
```
1. ./cv32/tb/uvmt_cv32/uvmt_cv32_pkg.sv:   import uvme_cv32_pkg::*;
2. ./cv32/tb/uvmt_cv32/uvmt_cv32_tb.sv:    import uvme_cv32_pkg::*;
```
And here is where `uvmt_cv32_pkg` is imported:
```
1. ./cv32/env/uvme_cv32/uvme_cv32_pkg.sv:   import uvmt_cv32_pkg ::*;
2. ./cv32/tb/uvmt_cv32/uvmt_cv32_tb.sv:     import uvmt_cv32_pkg::*;
```
Line #1 of both snipits shows a circular import reference where uvmt_cv32_pkg imports uvme_cv32_pkg and uvme_cv32pkg imports uvmt_cv32_pkg.

As far as I know, this is legal in SystemVerilog, and Metrics dsim compiles it without complaint.  However, it seems to give Questa trouble.  So my question to @datum-dpoulin is whether this is an intentional part of the UVM templates you generated.

datum-dpoulin, 2020-02-23:
> 1. ./cv32/env/uvme_cv32/uvme_cv32_pkg.sv:   import uvmt_cv32_pkg ::*;
This is indeed incorrect and should be removed. The original commit from the templates did not have this.

It was added by Mike (points odious finger) during commit b5f6040966b6371e357dd1755cad43df0f2c4d44

MikeOpenHWGroup, 2020-02-24:
I _knew_ that Mike guy could not be trusted.  Apologies for creating this churn.

I have removed the import of `uvmt_cv32_pkg ::*` in `./cv32/env/uvme_cv32/uvme_cv32_pkg.sv` and it compiles cleanly with DSIM.  Its on branch ohw_mike_20200223.   I will not merge it to the master since I want to be able to illustrate the issue to Metrics.  It seems to me that either Metrics DSIM or Mentor Questa has a compile issue and I'd like to get Metrics' opinion on this.  (In either case, we need to go with what works for both.)

@sjthales, please feel free to remove `import uvmt_cv32_pkg ::*` in `./cv32/env/uvme_cv32/uvme_cv32_pkg.sv` in your branch.

sjthales, 2020-02-24:
Ok thanks a lot for yours replies.

sjthales, 2020-02-24:
I commented import::uvmt_cv32_pkg*;  in uvme_cv32_pkg.sv file. The error disappears.

However, there are others errors.

In the file below, I encounter a problem , questa generates errors because clk_gen_vif is virtual.

cv32/tests/uvmt_cv32/base-tests/uvmt_cv32_base_test.sv
255 :   @(posedge clk_gen_vif.core_clock);
   @(posedge clk_gen_vif.core_reset_n);
   repeat (2) @(posedge clk_gen_vif.core_clock);

** Error (suppressible): /Work_dir/USE_CASE/2020/ohg/core-v-verif/cv32/sim/uvmt_cv32/../../../cv32/tb/uvmt_cv32/../../tests/uvmt_cv32/base-tests/uvmt_cv32_base_test.sv(255): (vlog-13262) A virtual interface element is not allowed in a sensitivity list.
** Error (suppressible): /Work_dir/USE_CASE/2020/ohg/core-v-verif/cv32/sim/uvmt_cv32/../../../cv32/tb/uvmt_cv32/../../tests/uvmt_cv32/base-tests/uvmt_cv32_base_test.sv(256): (vlog-13262) A virtual interface element is not allowed in a sensitivity list.
** Error (suppressible): /Work_dir/USE_CASE/2020/ohg/core-v-verif/cv32/sim/uvmt_cv32/../../../cv32/tb/uvmt_cv32/../../tests/uvmt_cv32/base-tests/uvmt_cv32_base_test.sv(257): (vlog-13262) A virtual interface element is not allowed in a sensitivity list.

In the same file, there are the errors below because uvmt_cv32_tb is not defined.

** Error (suppressible): /Work_dir/USE_CASE/2020/ohg/core-v-verif/cv32/sim/uvmt_cv32/../../../cv32/tb/uvmt_cv32/../../tests/uvmt_cv32/base-tests/uvmt_cv32_base_test.sv(298): (vlog-7027) The name ('uvmt_cv32_tb.tf') was not found in the current scope.  Please verify the spelling of the name 'uvmt_cv32_tb.tf'.
** Error (suppressible): /Work_dir/USE_CASE/2020/ohg/core-v-verif/cv32/sim/uvmt_cv32/../../../cv32/tb/uvmt_cv32/../../tests/uvmt_cv32/base-tests/uvmt_cv32_base_test.sv(299): (vlog-7027) The name ('uvmt_cv32_tb.tp') was not found in the current scope.  Please verify the spelling of the name 'uvmt_cv32_tb.tp'.
** Error (suppressible): /Work_dir/USE_CASE/2020/ohg/core-v-verif/cv32/sim/uvmt_cv32/../../../cv32/tb/uvmt_cv32/../../tests/uvmt_cv32/base-tests/uvmt_cv32_base_test.sv(300): (vlog-7027) The name ('uvmt_cv32_tb.evalid') was not found in the current scope.  Please verify the spelling of the name 'uvmt_cv32_tb.evalid'.

I encounter same kind of problem in cv32/tests/uvmt_cv32/compliance-tests/uvmt_cv32_firmware_test.sv.

** Error (suppressible): /Work_dir/USE_CASE/2020/ohg/core-v-verif/cv32/sim/uvmt_cv32/../../../cv32/tb/uvmt_cv32/../../tests/uvmt_cv32/compliance-tests/uvmt_cv32_firmware_test.sv(97): (vlog-7027) The name ('uvmt_cv32_tb.dut_wrap.ram_i.dp_ram_i.mem') was not found in the current scope.  Please verify the spelling of the name 'uvmt_cv32_tb.dut_wrap.ram_i.dp_ram_i.mem'.
** Error (suppressible): /Work_dir/USE_CASE/2020/ohg/core-v-verif/cv32/sim/uvmt_cv32/../../../cv32/tb/uvmt_cv32/../../tests/uvmt_cv32/compliance-tests/uvmt_cv32_firmware_test.sv(112): (vlog-13262) A virtual interface element is not allowed in a sensitivity list.
** Error (suppressible): /Work_dir/USE_CASE/2020/ohg/core-v-verif/cv32/sim/uvmt_cv32/../../../cv32/tb/uvmt_cv32/../../tests/uvmt_cv32/compliance-tests/uvmt_cv32_firmware_test.sv(113): (vlog-13262) A virtual interface element is not allowed in a sensitivity list.
** Error (suppressible): /Work_dir/USE_CASE/2020/ohg/core-v-verif/cv32/sim/uvmt_cv32/../../../cv32/tb/uvmt_cv32/../../tests/uvmt_cv32/compliance-tests/uvmt_cv32_firmware_test.sv(122): (vlog-13262) A virtual interface element is not allowed in a sensitivity list.
** Error (suppressible): /Work_dir/USE_CASE/2020/ohg/core-v-verif/cv32/sim/uvmt_cv32/../../../cv32/tb/uvmt_cv32/../../tests/uvmt_cv32/compliance-tests/uvmt_cv32_firmware_test.sv(123): (vlog-7027) The name ('uvmt_cv32_tb.evalue') was not found in the current scope.  Please verify the spelling of the name 'uvmt_cv32_tb.evalue'.

All these error are suppressibles, I need to add  -suppress 7027 -suppress 13262 in compilation options. However I think if questa generates some errors that means there is a problem in the code.  I think these codes need to be corrected.

MikeOpenHWGroup, 2020-02-24:
Hi @sjthales, thanks for the update.   I'll work with Metrics and investigate these issues.

sjthales, 2020-02-28:
Hello I succeeded in running the UVM testbench.
However to do so:

-  I added -suppress 13262 in command line to compile. That removes the vlog - 13262.

- I had to comment all cases where  there were vlog-7027 error. To add -suppress 7027 in compilation command line does not remove this error.  I always think there is bug. I have not find a way to correct that. 
- There is a problem of circular dependency between a module and a package.
If you use a variable (e.g. /mytop/mymodule/mysignal) of another module (e.g. module_b) in your module (e.g. module_a), QuestaSim requires that this module has been compiled previously (module_b before module_a).
Otherwise, as the variable does not exist in the work library, QuestaSim considers that there is an error (i.e. use of an not existing variable).

[image]
[image]

- at last, we encountered a problem with boot_addr definition in uvmt_cv32_clk_gen_if.sv file. boot_addr is defined on 8 bits like "output logic [8:0] boot_addr", but it is connected to the boot_addr_i signal of riscv_core module which is defined on 32 bits like   "input  logic [31:0] boot_addr_i". As a consequence, during simulation boot_addr_i was equal to 32'hZZZZZZ80, simulation on questa didn't work. To overcome the problem, we defined output logic [31:0] boot_add in uvmt_cv32_clk_gen_if.sv file.

 I am not an expert in UVM methodology but I think the circular dependency should be corrected. Questasim doesn't simulate such as UVM methodology is implemented today.

aimeepsutton, 2020-02-28:
Hello Sébastien,

My apologies for the delay in reply! Regarding the issues you've described above:

1. **Circular imports of packages** (uvmt_cv32_pkg.sv <-> imports <-> uvme_cv32_pkg.sv) should be removed. The Metrics simulator has been updated to flag this as an error in the future.
2. **vlog error 13262**: the error appears to be incorrectly firing on valid code. The code in question is not a sensitivity list, it is an event expression. If you consult section "25.9 Virtual Interfaces" or the SystemVerilog LRM (P1800-2012) you will find an example that matches this usage of a virtual interface member.
3. **vlog error 7027**: it concerns me that you had to comment out code to avoid this. My opinion is that the code in question is all legal. The references to "uvmt_cv32_tb.tf" etc. are hierarchical references to the top level of the testbench. In other testbenches/environments the top level module may have a more obvious name, such as "top" or "tb_top" but in this case "uvmt_cv32_tb" is the top of the simulation hierarchy, and it is legal to refer to it by the name of the module (since the top level module is not instantiated). It is also legal to make a forward reference to a module that has not yet been declared or compiled. My colleagues and I at Metrics have had experience with all major SystemVerilog simulators, and we believe this feature is supported by all of them. Perhaps Questa is not able to automatically determine that "uvmt_cv32_tb" is the top of the simulation hierarchy. Is there a command line argument you can pass to the simulator to specify this?  (something like "-top"?)
4. **Circular dependency described in your last comment**
 * my opinion is that no circular dependency exists. The line in question from "uvmt_cv32_firmware_test.sv" is:
```$readmemh(firmware, uvmt_cv32_tb.dut_wrap.ram_i.dp_ram_i.mem);```
The reference to uvmt_cv32_tb.dut_wrap... is a hierarchical reference to the top level module in the simulation hierarchy. This reference is legal and can be made before the top level module itself is defined or compiled. I believe that if we solve the issue above (vlog error 7027) we will also resolve this.

Mike and I have separately made some changes to the Makefile and one of the file lists, and these might improve the situation for you. Please apply your changes for Questa to Mike's latest branch and try again to compile and run a test. I will assign this issue back to Mike so he can tell you which branch to use for further testing.

I am happy to continue to work with you on these issues. Please let me know how it goes!

MikeOpenHWGroup, 2020-02-28:
All of the changes @aimeepsutton mentions here, plus several of my own for other items related to Makefiles, can be found on the `development` branch of https://github.com/openhwgroup/core-v-verif.

MikeOpenHWGroup, 2020-03-02:
Hi @sjthales, I've removed myself and @aimeepsutton from the "Assignee" list of this issue as I believe the ball is in your court.   Have a look at the `development` branch and let us know if you can move this issue forward.

agrasset, 2020-03-03:
Hi @aimeepsutton @MikeOpenHWGroup ,
I think that the hierarchical reference to the top level testbench that @sjthales describes in the "vlog error 7027" point is not legal. The reference is done in a file included in a package.
And according to the reference manual items within packages shall not have hierarchical references to identifiers except those created within the package.

saurabhsingh-pvips, 2020-04-15:
Is the master branch stable to run simulation using Questa? I am facing compilation error in UVM Package. See 
[questa_compile.log](https://github.com/openhwgroup/core-v-verif/files/4482161/questa_compile.log)

MikeOpenHWGroup, 2020-04-15:
Hi @saurabhsingh-pvips.   It is unlikely that the Questa Makefiles work.  There have been several updates to the UVM environment and our colleagues that support Questa have been off-line due to the on-going COVID-19 lockdowns.

I am unable to help as I lack access to Mentor tools.  I suggest having a look at `dsim.mk` or `xrun.mk` as these are both stable and running.

strichmo, 2020-09-08:
The Questa Makefiles are under my watch as the current Questa tool prime (but I would be happy to start someone else on this as I also am the Xcelium tool prime).  All Questa makefiles should be updated and supporting the common cv32e40p UVM testbench makefile targets and variables.

https://github.com/strichmo/core-v-verif/blob/master/cv32/sim/uvmt_cv32/README.md

MikeOpenHWGroup, 2020-10-16:
This task has been completed for some time and I have been delinquent in closing it.  Questa Makefiles are know to be working due to the efforts of @strichmo.  Closing now.

closed 2020-10-16 by MikeOpenHWGroup