openhwfoundation/core-v-verif #65: Integration of the existing tests in the CV32 repository

davideschiavone, 2020-02-24:
# Existing tests

The CV32E40P repository has a tb directory inside the repository that contains some tests as:

1. riscv compliance tests
2. interrupt tests
3. debug tests (dm)

https://github.com/openhwgroup/cv32e40p/tree/master/tb

We want to move all of these tests in this repository and leave there only an `Hello world` example.

### Task Outcome

All the tests from the CV32 repository are moved to this repository. After that, the core won't have any test program but only an simple application to let the user try out the core.

### Background information

Discussions on this topic available at

https://github.com/openhwgroup/cv32e40p/pull/238
https://github.com/openhwgroup/cv32e40p/issues/229
https://github.com/openhwgroup/cv32e40p/issues/208

MikeOpenHWGroup, 2020-02-25:
Thanks Davide.   I've added a `task` label to this issue and placed it in the `CV32E40P RTL + UVM Environment Integration` project.

The riscv compliance tests are already running on both the `core` testbench and `uvmt_cv32` environments in core-v-verif.   I'll start work on interrupt tests and debug tests starting on February 26.

MikeOpenHWGroup, 2020-03-02:
Update.  I'm ready to move the `interrupt` tests any time.  I'd like to wait for [pr 245](https://github.com/openhwgroup/cv32e40p/pull/245) to be complete as I have a rather large (and getting larger) set of changes on the [core-v-verif development branch](https://github.com/openhwgroup/core-v-verif/tree/development) I would like to merge to the master and this is gated by 245.

The `debug test` will take more time as we have not yet fully thought out our strategy wrt debug verification.

MikeOpenHWGroup, 2020-10-28:
Hi @davideschiavone, I would like to close this issue.  The debug and interrupt tests mentioned in this issue have been superseded by a set of new and more extensive [test-programs specifically written for cv32e40p](https://github.com/openhwgroup/core-v-verif/tree/master/cv32/tests/programs/custom), plus a set of [pseudo-random tests](https://github.com/openhwgroup/core-v-verif/tree/master/cv32/tests/programs/corev-dv) that are each run 10s of times per regression.

It is not certain that the cv32e40p tests in the [example_tb](https://github.com/openhwgroup/cv32e40p/tree/master/example_tb) would work in the core-v-verif environment.

Silabs-ArjanB, 2020-10-29:
Hi @MikeOpenHWGroup One remark on this: With respect to Debug there is a 'v013.2 Debug Compliance test' that I think would be very valuable to add to core-v-verif. It requires the instantiation of the Debug Module inside the core-v-verif testbench and it will then do a (basic) compliance check of the combination of the RISC-V core and the Debug Module. Maybe this is beyond the scope of the RTL freeze target, but if so, then I would really appreciate if this can be added afterwards as it represents actual usage scenarios of the debug system.

MikeOpenHWGroup, 2020-11-04:
As per my comments above and based on consensus with other Contributors, I am closing this issue.

closed 2020-11-04 by MikeOpenHWGroup