openhwfoundation/core-v-verif #66: Update cv32/sim/uvmt_cv32/Makefile for Cadence Xcelium support

MikeOpenHWGroup, 2020-02-28:
## Task Outcome
On the **_development_** branch of [core-v-verif](https://github.com/openhwgroup/core-v-verif) update the Makefile in `cv32/sim/uvmt_cv32` to add/update the **xrun** targets to match the **dsim** targets.  Below is a complete list:
* **make xrun-no-firmware UVM_TESTNAME=uvmt_cv32_smoke_test**:compile, run, come of out reset and die.
* **make xrun-hello-world**: run the hello_world program found at `../../tests/core/custom`.
* **make dsim-cv32-riscv-tests**: run the CV32-specific RISC-V tests found at `../../tests/core/cv32_riscv_tests_firmware`
* **make xrun-cv32-riscv-compilance-tests**: run the CV32-specific RISC-V tests found at `../../tests/core/cv32_riscv_compliance_tests_firmware`
* **make xrun-firmware**: run all the programs found at `../../tests/core/firmware`.
* **make xrun-riscv-tests**: run the RISC-V tests found at `../../tests/core/riscv_tests`
* **make xrun-riscv-compilance-tests**: run the RISC-V tests found at `../../tests/core/riscv_compliance_tests`
* **make xrun-unit-test <prog>**: Run one <prog> from the firmware suite of tests.  For example: `make xrun-unit-test addi`

### Background information
There has been a great deal of churn in the organization of the [cv32e40p](https://github.com/openhwgroup/cv32e40p) and [core-v-verif](https://github.com/openhwgroup/core-v-verif) repositories.  This has had a significant impact on the Makefiles and its almost certain that the **_xrun_** target no longer work.

### Location Information
The Makefile in question is at `cv32/sim/Makefile`.  Not that this file includes `../Common.mk` and this included file may need to be updated as well.

### Completion Criteria
All of the above targets compile and run to completion.

### Additional Context
Please fork from the **development** branch, not the master branch.

alfredoh1234, 2020-03-02:
@MikeOpenHWGroup , I've ran:
"make xrun-hello_world", and "make xrun-firmware" with no issues with regards to the directory structure changes.

I am working on implementing the other Makefile updates right now

MikeOpenHWGroup, 2020-03-03:
Thanks @alfredoh1234.  Don't forget issue #61.

MikeOpenHWGroup, 2020-03-23:
This task is complete - closing,

closed 2020-03-23 by MikeOpenHWGroup