openhwfoundation/core-v-verif #87: Port CV32E40P Interrupts Tests from RTL repo to core-v-verif repo

MikeOpenHWGroup, 2020-03-30:
### Task Outcome
Port all existing interrupt testcases from the `tb/core/interrupt_test` directory in the `cv32e40p` repo to the `core-v-verif` repository.

### Background information
At the time of this writing (2020-03-30), there is very little documentation on the CV32E interrupts or the existing testcases.

### Location Information
Existing testcase code is located at https://github.com/openhwgroup/cv32e40p/tree/master/tb/core/interrupt_test

New location of the code should be https://github.com/openhwgroup/core-v-verif/tree/master/cv32/tests/core

### Completion Criteria
Tests run and pass in the `uvmt_cv32` environment.
Running under the `core` testbench is nice-to-have, but not a requirement.

MikeOpenHWGroup, 2020-06-29:
Update.  The first part of this task (move testcase from cv32e40p repo to core-v-verif) has been completed.

strichmo, 2020-09-08:
The test was ported to the cv32e40p according to directed test criteria from the interrupts vplan.  The test is complete and part of regressions.

MikeOpenHWGroup, 2020-09-09:
>  The test is complete and part of regressions.

Excellent @strichmo, thanks.  When you say "part of regressions", do you also mean in the .metrics.json control file for MetricsCI?

MikeOpenHWGroup, 2020-10-16:
This task has been completed for some time and I have been delinquent in closing it.  Closing now.

closed 2020-10-16 by MikeOpenHWGroup