openhwfoundation/core-v-verif #89: cannot find riscv_config.sv

eroom1966, 2020-03-31:
Hi Mike

@MikeOpenHWGroup 
@simon5656

Following your recent request to take a look at the REPO
I have tried following the instructions here cv32/sim/core/README 
I am usinq Questa and tried 

make firmware-vsim-run

but got the following error 

> ** Error: /home/moore/git/eroom1966/core-v-verif/cv32/sim/core/../../../cv32/tb/tb_riscv/riscv_random_stall.sv(25): (vlog-13006) Could not find the package (riscv_defines).  Design read will continue, but expect a cascade of errors after this failure.  Furthermore if you experience a vopt-7 error immediately before this error then please check the package names or the library search paths on the command line.
> 
> ** Error: /home/moore/git/eroom1966/core-v-verif/cv32/sim/core/../../../cv32/tb/tb_riscv/riscv_random_stall.sv(27): Cannot find `include file "riscv_config.sv" in directories:
>     , /home/nda/Mentor/Questa/install/10.7c_3/questasim/ovm-2.1.2/../verilog_src/ovm-2.1.2/src, /home/nda/Mentor/Questa/install/10.7c_3/questasim/uvm-1.1d/../verilog_src/uvm-1.1d/src
> 

I searched in the repo for riscv_config.sv, and I do find it here

> core-v-verif/core-v-cores/cv32e40p/rtl/include/riscv_config.sv

This was the command line executed, not sure if this could be missing an include path ?

> vlog -work work "+incdir+" -pedanticerrors -suppress 2577 -suppress 2583 /home/moore/git/eroom1966/core-v-verif/cv32/sim/core/../../../cv32/tb/tb_riscv/include/perturbation_defines.sv /home/moore/git/eroom1966/core-v-verif/cv32/sim/core/../../../cv32/tb/core/tb_top.sv /home/moore/git/eroom1966/core-v-verif/cv32/sim/core/../../../cv32/tb/core/riscv_wrapper.sv /home/moore/git/eroom1966/core-v-verif/cv32/sim/core/../../../cv32/tb/core/mm_ram.sv /home/moore/git/eroom1966/core-v-verif/cv32/sim/core/../../../cv32/tb/core/dp_ram.sv /home/moore/git/eroom1966/core-v-verif/cv32/sim/core/../../../cv32/tb/tb_riscv/riscv_random_stall.sv /home/moore/git/eroom1966/core-v-verif/cv32/sim/core/../../../cv32/tb/tb_riscv/riscv_random_interrupt_generator.sv

MikeOpenHWGroup, 2020-03-31:
Hi @eroom1966.  Yes, it is known that the Questa support is broken (I reached out to Simon to have a look at it last week).   I'd recommend starting out with the UVM environment, not the core testbench.

I'm in a concall starting in 5 minutes, but will have a look at this later this morning (NA EDT).

eroom1966, 2020-03-31:
Hi @MikeOpenHWGroup 
Great, thanks.
Lee

MikeOpenHWGroup, 2020-04-27:
Hi Lee (@eroom1966), I assume this issue can be closed now, is that correct?

PauliusMorku, 2020-05-31:
I have the same issue, is there a known solution for this?

MikeOpenHWGroup, 2020-05-31:
Update.  The "firmware" targets are in the process of being deprecated.  As such, they should be removed form the vsim.mk file.

I do not have access to Questa, so cannot do this work myself.  @eroom1966 , @sjthales can either of you look into this?

MikeOpenHWGroup, 2020-05-31:
Hi @PauliusMorku, thanks for adding to this issue.  Can you add an update with the specific commands you are running?   Please include the set-up sets specifying any special shell env variables and the directory you are invoke `make` from.   Thanks!

PauliusMorku, 2020-05-31:
So I invoked
`core-v-verif/cv32/sim/core$ make questa-unit-test addi`
(tried to target firmware-vsim-run and questa-unit-test I_LBU_01 with same error)
The output:
`-- Compiling module dp_ram
** Error: /home/pmorku/workspace/core-v-verif/cv32/sim/core/../../../cv32/tb/core/tb_riscv/riscv_random_stall.sv(25): (vlog-13006) Could not find the package (riscv_defines).  Design read will continue, but expect a cascade of errors after this failure.  Furthermore if you experience a vopt-7 error immediately before this error then please check the package names or the library search paths on the command line.
** Error: /home/pmorku/workspace/core-v-verif/cv32/sim/core/../../../cv32/tb/core/tb_riscv/riscv_random_stall.sv(27): Cannot find `include file "riscv_config.sv" in directories:
    , /home/pmorku/software/questasim_10.7c/questasim/ovm-2.1.2/../verilog_src/ovm-2.1.2/src, /home/pmorku/software/questasim_10.7c/questasim/uvm-1.1d/../verilog_src/uvm-1.1d/src
** Error: /home/pmorku/workspace/core-v-verif/cv32/sim/core/../../../cv32/tb/core/tb_riscv/riscv_random_interrupt_generator.sv(26): (vlog-13006) Could not find the package (riscv_defines).  Design read will continue, but expect a cascade of errors after this failure.  Furthermore if you experience a vopt-7 error immediately before this error then please check the package names or the library search paths on the command line.
** Error: /home/pmorku/workspace/core-v-verif/cv32/sim/core/../../../cv32/tb/core/tb_riscv/riscv_random_interrupt_generator.sv(28): Cannot find `include file "riscv_config.sv" in directories:
    , /home/pmorku/software/questasim_10.7c/questasim/ovm-2.1.2/../verilog_src/ovm-2.1.2/src, /home/pmorku/software/questasim_10.7c/questasim/uvm-1.1d/../verilog_src/uvm-1.1d/src
End time: 23:40:34 on May 31,2020, Elapsed time: 0:00:00
Errors: 4, Warnings: 0
make: *** [Makefile:181: .build-rtl] Error 2
`

MikeOpenHWGroup, 2020-05-31:
Ugh.  I see that the makefile is very out of date as far as Questa is concerned.

@sjthales can you have a look at this?

sjthales, 2020-06-02:
@MikeOpenHWGroup @PauliusMorku @eroom1966 I found out why it doesn't work.
Some lines for Questa compilation have been commented for [https://github.com/openhwgroup/core-v-verif/commit/2492948fdd9c19a471f0c5aa637ac54fde942999](url).

You just have to uncomment the lines below in the makefile and that will work. 

`#.lib-rtl:`
`#	$(VLIB) $(VWORK)`
`#	touch .lib-rtl`

`#.build-rtl: .lib-rtl $(CV32E40P_PKG) $(TBSRC_PKG) $(TBSRC)`
`#	$(VLOG) \`
`#		-work $(VWORK) \`
`#		$(VLOG_FLAGS) \`
`#		-f $(CV32E40P_MANIFEST) \`
`#		$(TBSRC_PKG) $(TBSRC) `
		

`#vsim-all:  .opt-rtl`
	

`#.opt-rtl: .build-rtl`
`#	$(VOPT) -work $(VWORK) $(VOPT_FLAGS) $(RTLSRC_VLOG_TB_TOP) -o` `$(RTLSRC_VOPT_TB_TOP)`
`#	touch .opt-rtl`

PauliusMorku, 2020-06-03:
I can confirm that the error is gone. However, firmware-vsim-run still doesn't work, but as I understood it is deprecated. I also had to add  -suppress 2732 and -suppress 2912 to VLOG_FLAGS in order to make it work. Anyway, thank you for the help.

sjthales, 2020-06-03:
Actually, I noticed that .lib-rtl, .build-rtl, .opt-rtl and vsim-all targets were already defined in the Makefile, I need to dig deeper to find the problem before submitting a pull-request.

MikeOpenHWGroup, 2020-06-03:
Thanks @PauliusMorku and @sjthales.   I'll wait for Sebastien's pull-request.  Please consider removing the `firmware` target(s) as part of that.

sjthales, 2020-06-04:
Done @MikeOpenHWGroup , @PauliusMorku firmware-vsim-run target was actually obsolete, questa-firmware replaces it.

MikeOpenHWGroup, 2020-06-04:
Thanks @PauliusMorku and @sjthales.   I've merged in PR #89 and also updated the README in cv32/sim/uvmt_cv32 to better reflect the status of the Makefiles.  Much appreciated.

Hi Lee (@eroom1966).   As you were the original reporter of this issue, I have assigned it back to you.  If you are content with the fixes, please close this issue.  Otherwise, re-assign it back to me.

MikeOpenHWGroup, 2020-06-29:
Hi Lee (@eroom1966).   Can you have a look at this?   Thanks.

MikeOpenHWGroup, 2020-10-24:
As this issue appears to be resolved and there has been no comment for four months, I will close it.

closed 2020-10-24 by MikeOpenHWGroup