openhwfoundation/core-v-verif #105: Clarify PULP RISCV GNU Toolchain configuration for cv32/sim/core/

Smattacus, 2020-05-11:
Several configuration options are available for the [PULP toolchain](https://github.com/pulp-platform/pulp-riscv-gnu-toolchain) (Pulp, newlib, linux, linux multilib). It is not clear from the readme in [cv32/sim/core](https://github.com/openhwgroup/core-v-verif/tree/master/cv32/sim/core) which option to install, and installing the wrong one results in compiler errors while running Verilator make targets. The correct one was the Newlib cross-compiler for all Pulp variants - Installation (Pulp).

Until a dedicated core-v toolchain is defined and available, a proposed containment is to update the readme to reference the Installation (Pulp) section.

I'll submit a PR on this soon.

MikeOpenHWGroup, 2020-05-11:
This is a great suggestion @Smattacus.  As you may be aware, the OpenHW Group has created a new Task Group.  One of the co-chairs of this TG, @jeremybennett is driving an effort to standardize the toolchain.  Let's hold off on the PR until Jeremy has had a chance to chime in.   In the meantime, we can use this issue to record comments.

jeremybennett, 2020-05-15:
@Smattacus We'd appreciate your input to the SW task group. One thing I hope we can sort out sooner rather than later is simple binary downloads (similar to those for generic RISC-V at https://www.embecosm.com/resources/tool-chain-downloads/). The majority of users of CORE-V will not want to learn how to build a full compiler tool chain from source!

MikeOpenHWGroup, 2020-06-29:
Hi @Smattacus.   Can this issue be closed now?  The VTG has no plans to take action on this as we are deferring to (indeed: _relying on_) the SWTG to define the toolchain we will use.

MikeOpenHWGroup, 2020-10-19:
I have recommended that this issue be closed and there has been no comment on this issue for > 3 months, so I am closing now.

closed 2020-10-19 by MikeOpenHWGroup