openhwfoundation/core-v-verif #127: Question about pseudo instructions in the coverage model

MikeOpenHWGroup, 2020-06-17:
Hi Lee (@eroom1966).   A question came up when reviewing the functional coverage model you donated with @alfredoh1234.  I'm writing it up as a GitHub issue as this discussion strikes me as one worth preserving.

Alfredo noticed several RV32I instructions in the model that are not explicitly called out in our [RV32I Vplan](https://github.com/openhwgroup/core-v-docs/blob/master/verif/CV32E40P/VerificationPlan/base_instruction_set/CV32E40P_RV32I_Base_Instructions.xlsx).  A quick glance leads me to believe that these are pseudo-instructions as per Table 26.2 of the base ISA.  This leads to a number of questions:

1. Are pseudo-instructions part of the coverage model?
2. How does the ISS identify pseudo-instructions?
3. In your opinion, is there any value, _from an RTL verification perspective_ having functional coverage for pseudo instructions in the coverage model?

The coverage model in question is located at https://github.com/openhwgroup/core-v-verif/blob/master/cv32/tb/uvmt_cv32/class_coverage.svh

eroom1966, 2020-06-17:
Hi @MikeOpenHWGroup @alfredoh1234 
This is a great question that we have discussed many times internally

The problem with pseudo instructions is that they overlap with instructions defined as part of the ISA, and even more complicated than that, some pseudo instructions do not have a single mapping, for instance the 'li' (load immediate) or the 'la' (load address)  instructions, map across to multiple individual instructions.
I know Simon @simon5656 has discussed this more often than myself, I have pinged him to get his thoughts on this matter also.

As far as the RISCV OVP models are concerned, they will execute the instructions from there exact bit encodings. the disassembler, will additionally use those bit encodings to determine if they pattern match a particular pseudo instruction. As far as possible the model disassembler, attempts to match the binutils objdump command. The exception to the pseudo matches is clearly the mapping of pseudo instructions to multiple instructions (li, la), that is definitely not going to produce a disassembly output - nor will objdump attempt to provide disassembly of those instructions.

Thx
Lee

alfredoh1234, 2020-06-17:
@eroom1966 and @simon5656 ,
I have a question related to this, but not from a pseudo-instruction perspective. From an RTL (hardware implementation of instruction decoder) :
* is it more valuable to make sure instruction exceptions are robustly/consistently handled? versus covering all pseudo-instruction mappings?

My thinking is that from a hardware testing perspective, I will not have control over how the assembler maps pseudo-instructions onto ISA instructions; but we can verify that anytime there is a mapping outside the supported ISA, that will be handled correctly. 

***
Alfredo

simon5656, 2020-06-17:
@MikeOpenHWGroup, @eroom1966  good question on pseudos

I believe that for HW DV there is no purpose in the measurement of functional coverage of pseudo instructions - they are an assembler convenience.

If there are pseudos in the Imperas provided SystemVerilog functional coverage code - we can provide a more up to date version that does not include any pseudos.

[fyi - in our internal coverage tools - we do have switches so we can do coverage on them - but this to assess tool chains etc.]

thanks
Simon

MikeOpenHWGroup, 2020-06-17:
Thanks @simon5656.   I'm not too worried about having a functional coverage model with pseudo-instructions.  These are easy enough for us to remove (still not decided that we will).  My worry is about what the reference model's disassembler feeds the coverage model.  If a physical instruction is disassembled into a matching pseudo-instruction, then our coverage of physical instructions may have holes.   We would want the ability to disable disassembly into pseudo-instructions so that our functional coverage tracks the actual instructions the RTL processes.

eroom1966, 2020-06-17:
@alfredoh1234 
As far as the reference is concerned it is absolutely Instruction set accurate from the Standard RISCV ISA Extensions. This is very important especially from the perspective of the question _I think_ you are asking, which is what happens in the illegal conditions.
Under illegal instruction conditions, or the usage of un-supported instructions via the subsets, then the reference will behave accordingly in terms of its trap/exception behavior
- did I understand what you were asking correctly ?
thx
Lee

MikeOpenHWGroup, 2020-06-18:
Hi @eroom1966, I also think you addressed @alfredoh1234, question - thanks.

The discussion has strayed from the original questions:

`1. Are pseudo-instructions part of the coverage model?`
The answer is yes.

`2. How does the ISS identify pseudo-instructions?`
This was answered to my satisfaction (above).  Thanks Lee.

`3. Is there any value of having functional coverage for pseudo instructions in the coverage model?`
As is often the case in verification, opinions differ.  In such circumstances I reserve the deciding vote and therefore we will _not_ implement functional coverage of pseudo-instructions since the RTL never sees them.

Given that we have answers to the three questions posed by this issue, I will close it.  Thanks all.

closed 2020-06-18 by MikeOpenHWGroup