openhwfoundation/core-v-verif #132: Fail to run core testbench Hello_World

saumeister, 2020-06-25:
Hi everyone,

I struggle to run the hello world in the core test-bench provided and are looking for some pointers to start using it. 

I started to uncomment the hello_world example in the common.mk file and proceeded to to try:
 `make questa-hello_world`
This triggers three errors because the crt0.s,syscalls.c and vector.s files can't be found in the custom folder.

I then tried to solve this by using the provided files in the bsp folder instead, inducing the linker
script. This created other issues with several undefined references in the vector.S file
such as sw_irq_handler, __no_irq_handler etc. Does anyone have any pointers how I should proceed to
get it running?

Best Regards
Saumeister

MikeOpenHWGroup, 2020-06-25:
Hi @saumeister.  I am not suprised to hear that make `questa-hello_world` is not working.   As you may know, OpenHW relies on its members to maintain the Makefiles for the various SV simulators and I am constantly surprised that there are few Questa users on this project.  I myself do not have access to Questa so cannot directly assist you.

I can report that the Makefiles for the `core` testbench and `uvmt_cv32` verification environment are sane and working for Metrics DSIM and Cadence Xcelium (XRUN).   The `core` testbench also works using Verilator (note that the UVM environment does not run with Verilator).

So, if you are using Verilator, the following will work:
```
$ git clone https://github.com/openhwgroup/core-v-verif
$ cd core-v-verif/cv32/sim/core
$ make
```
For dsim you can use `make dsim-hello_world` and for Xcelium `make xrun-hello_world`.

Please have a look at cv32/sim/core/Makefile.  The rules/targets used by Verilator/dsim/xrun should be simple to replicate for Questa.   Once you get it working, please submit a pull-request.

Lastly, please note that the `core` testbench is not our "main" testbench and is there as an example only.  You should be using the UVM verification environment, `uvmt_cv32`.

Thanks!

saumeister, 2020-06-25:
Hi, @MikeOpenHWGroup thanks for the thorough answer! I'm actually merging to Xcelium tomorrow so nice to hear it is up to date. I'm loosing access to Questasim so unfortunately I do not have time to update the files since it will take some time with my current low skill level regarding this. I will give it a try and use the 'uvmt_cv32' test-bench, I was initially testing the 'core' test-bench as I previously worked with the one provided in the core repo. 

Br
Saumeister

alfredoh1234, 2020-06-26:
on the latest version on the master branch: 
   make SIMULATOR=xrun hello_world

It seems like the target names have changed

MikeOpenHWGroup, 2020-06-26:
Hi @alfredoh1234.  Yes there have been (and will continue to be) lots of changes as the project is in active development.   It's a gy idea to update your clone at least daily and track updates to the READMEs.

@saumeister, can we close this issue now?

saumeister, 2020-06-28:
I will close the issue now. Thanks for the help, both of you!

closed 2020-06-28 by saumeister