openhwfoundation/core-v-verif #142: Make targets from https://github.com/openhwgroup/core-v-verif/tree/master/cv32/sim/uvmt_cv32 not working

Silabs-ArjanB, 2020-07-07:
Some of the make targets listed on https://github.com/openhwgroup/core-v-verif/tree/master/cv32/sim/uvmt_cv32 do not work for the 040ae4eac461c76f1d302a9ffcb28b34c3300b0a repos of core-v-verif:

[image]

- make hello-world

Works

- make cv32-riscv-tests

make: *** No rule to make target `/home/arbink/openhw/master/core-v-verif/cv32/sim/uvmt_cv32/../../../cv32/tests/core/cv32_riscv_tests_firmware/cv32_riscv_tests_firmware.hex', needed by `cv32-riscv-tests'.  Stop.

- make cv32-riscv-compilance-tests

Obvious typo in target, but after trying 'make cv32-riscv-compliance-tests' the test failsas follows:

  # Test part A5 - test base address + 2047
  out of bounds read from fffffacb
  xmsim: *F,FATSEV (/home/arbink/openhw/master/core-v-verif/cv32/tb/core/mm_ram.sv,360): (time 448876500 PS).
  uvmt_cv32_tb.dut_wrap.ram_i.read_mux

- make make dsim-unit-test

Double 'make'

There is supposed to be a 'new' Compliance suite as well: make cv32-new-riscv-compliance-all however also fails with a 'no rule to make target.

All above is done with SIMULATOR=xrun.

eroom1966, 2020-07-07:
I am also having trouble, but the issue I see is around the cross compiler.
The cross compiler I am using is no longer producing the verilog memory file in the correct format required as part of the flow.
I can only imagine the linker script has caused a change.

I tried rebuilding the toolchain from scratch using instructions from https://github.com/riscv/riscv-gnu-toolchain, now I am getting lots of errors of the sort
`ABI is incompatible with that of the selected emulation:`
I cannot find any instructions about how to build (the expected) toolchain for this flow, can anybody provide a pointer ?

saumeister, 2020-07-08:
Hi @eroom1966,

They recently added Toolchain Installation Instructions at https://github.com/openhwgroup/core-v-verif/blob/master/cv32/sim/TOOLCHAIN.md . I have not tried it running that exact code but did something similar myself but hopefully is provides you some guidance.

strichmo, 2020-09-08:
The recent simulation control makefile updates are defining proper targets for the Xcelium makefile to deal with general "custom" tests and "corev-dv" tests.

Running compliance outside of dsim is still in development.

MikeOpenHWGroup, 2020-10-28:
Hi @Silabs-ArjanB, this issue has been idle for some time.  Is this still an issue for you?

Silabs-ArjanB, 2020-10-29:
Hi @MikeOpenHWGroup I see that the information provided on  https://github.com/openhwgroup/core-v-verif/tree/master/cv32/sim/uvmt_cv32 has been fixed now, thank you. This issue in itself can therefore be closed.

Still we have the known issue of signature mismatches during the compliance tests. You told me these are false warnings that should be ignored; that is fine for now. I am still expecting that the issue will eventually be fixed and we track that via https://github.com/openhwgroup/core-v-verif/issues/253.

I also notice that the Compliance suite is not being run for rv32Zicsr and rv32Zifencei. I will open a feature request for that as I think we should always run such available test cases.

Silabs-ArjanB, 2020-10-29:
Hi @MikeOpenHWGroup I closed this too early. The following example command is from https://github.com/openhwgroup/core-v-verif/tree/master/cv32/sim/uvmt_cv32 is not recognized:

make compliance RISCV_ISA=rv32Zifencei COMPLIANCE_PROG=I-FENCE.I-01

How can I run a specific compliance test stand-alone?

MikeOpenHWGroup, 2020-11-02:
@Silabs-ArjanB, please check the latest [README](https://github.com/openhwgroup/core-v-verif/tree/master/cv32/sim/uvmt_cv32#risc-v-compliance-test-suite), and let me know that answers your questions.  I have tested these instructions using both dsim and xrun.

Note that we do not have a specific target that runs all tests in the compliance suite, although we do have a specific regression (cv32_static_compliance_regression) for it.

closed 2020-11-02 by Silabs-ArjanB