openhwfoundation/core-v-verif #147: running assembly code through Cadence (Xcelium) ?

osta1, 2020-07-09:
Is it possible to run assembly codes (riscv_ tests and riscv_compliance_tests) through Cadence (xcelium) in the "sim"directory? Usually a c code could be run through the following command : make xrun-hello_world. 
Is there any step required in case of running assembly programs instead of c program , or some modifications should be added to the makefile?

MikeOpenHWGroup, 2020-07-09:
Hi @osta1.  The short answer to your question is yes.  Most of the test-programs used in core-v-verif for the CV32E40P are assembly, either manually written or generated by the Google [riscv-dv](https://github.com/google/riscv-dv) RISC-V instruction stream generator.

Probably the easiest thing to do is place the assembly source files in cv32/tests/core/custom and use the `custom` target of the makefile to compile/run it:
```
make custom SIMULATOR=xrun CUSTOM_PROG=<my_assembly_program>
```
Note that `<my_assembly_program>` should _not_ have an extension in the above command line.

Before trying this you should familiarize yourself with the testbench and verification environment.  The [Verification Strategy](https://core-v-docs-verif-strat.readthedocs.io/en/latest/) document will have lots of good information for you.  Also, if you are writting your own assembly programs for this environment, you will need to make your code run with the testbenches [Virtual Peripherals](https://core-v-docs-verif-strat.readthedocs.io/en/latest/sim_tests.html#virtual-peripherals) and use the resources of the [Board Support Package](https://github.com/openhwgroup/core-v-verif/blob/master/cv32/bsp/README.md).  In addition to these resources, there are READMEs through-out the repository to help you through it.

Its also important to have the right [toolchain](https://github.com/openhwgroup/core-v-verif/blob/master/cv32/sim/TOOLCHAIN.md) installed.  You may get unpredictable results otherwise and we cannot help you if that.

Let me know if this helps and if I can close this issue.   Thanks again.

osta1, 2020-07-09:
Thank you for your answer. I read most of the documentation before making some tests and I have successfully compiled some c code benchmarks. I have installed the pulp-riscv-gnu-toolchain which should be the right one. I tried to compile an assembly code through the command that you have mentioned; the error generated is (Verilator command not found). But I am using Xcelium simulator not the Verilator. 
Another question , How can I only invoke the assembler ? I just want to try some simple assembly codes.

Thanks for your help.

MikeOpenHWGroup, 2020-07-09:
I suspect that if you are seeing Verilator messages you are attempting to run from the `cv32/sim/core` directory.  That is a simple testbench that is there for historical reasons and is not fully supported.   You should run all your tests using the UVM environment from `cv32/sim/uvmt_cv32`.  Also, you will need to specify your simulator with either a shell or Makefile variable, SIMULATOR (see the example command I provided, above).

If you _must_ use the core testbench, understand that the structure of the Makefile is almost entirely different, and I do not think it supports the `custom` target.  The README in that directory should be (but is not guaranteed to be) up to date.

To invoke the assembler directly, you'll need to look at the targets in the Makefiles that produce ELFs.   We do not run these directly directly (and it been a long time since I looked at them).

MikeOpenHWGroup, 2020-07-15:
Hi @osta1, are your questions resolved?   If so, I would like to close this issue (we do not like to leave issues in an unresolved state for long).   If not, please provide an update and we'll see what we can do to help you resolve it.   If we do not hear from you in a week, this issue will be manually closed.

osta1, 2020-07-15:
Hi Mike, thank you for your answers. I have succeeded in running the assembly programs without going to the uvm directory; only by running the following command: “make xrun-firmware”. The problem is resolved, I will close the issue. 
Thank you.

closed 2020-07-15 by osta1