openhwfoundation/core-v-verif #156: FCOV for RV32C

MikeOpenHWGroup, 2020-07-14:
### Problem Description
The current functional coverage model for RV32C instructions has two issues:
1. It does not collect any coverage at all.
2. It relies on the Reference Model (OVPsim ISS) to disassemble machine code into compressed instruction mnemonics.

### Root Causes
1. The constructor (`new()`) does not construct the cover_groups for the RV32C instructions.   I have taken the liberty of fixing this as below:
```
    function new();
       add_cg = new();
       addi_cg = new();
       and_cg = new();
       ... 
       remu_cg = new();
       c_add_cg = new();
       c_addi_cg = new();
       c_addi16sp_cg = new();
       c_addi4spn_cg = new();
       // etc...
    endfunction: new
```

2. As we discussed via email, the RM does not supply mnemonics for RV32C instructions.  So we need to build that up ourselves in the coverage model.   I have prototyped something that seems to work for c.add, c.addi, c.addi16sp and c.addi4spn.   It is a two step process:

```
// Step One: identify compressed instructions
//                  the function check_compressed() is called by sample()
    function void check_compressed(input ins_t ins);
        case (ins.ins_str)
            "add"      : if ( get_gpr_name(ins.ops[0].val, ins.ops[0].key, "add")  == get_gpr_name(ins.ops[1].val, ins.ops[1].key, "add") ) c_add_cg.sample(ins);

            "addi"     : if ( get_gpr_name(ins.ops[0].val, ins.ops[0].key, "addi") == get_gpr_name(ins.ops[1].val, ins.ops[1].key, "addi")) c_addi_cg.sample(ins);

            "addi16sp" : if ( get_gpr_name(ins.ops[0].val, ins.ops[0].key, "addi") == get_gpr_name(ins.ops[1].val, ins.ops[1].key, "addi") && 
                              get_gpr_name(ins.ops[0].val, ins.ops[0].key, "addi") == "x2") c_addi16sp_cg.sample(ins);

            "addi4spn" : if ( get_gpr_name(ins.ops[0].val, ins.ops[0].key, "addi") == get_gpr_name(ins.ops[1].val, ins.ops[1].key, "addi") && 
                              get_gpr_name(ins.ops[1].val, ins.ops[1].key, "addi") == "x2") c_addi4spn_cg.sample(ins);
 
            default: `uvm_info("RV32ISA Coverage", $sformatf("check_compressed(): ins [%0s] not yet checked", ins.ins_str), UVM_HIGH)
        endcase
    endfunction: check_compressed

// Step Two: sample compressed instructions
    covergroup c_addi_cg with function sample(ins_t ins);
        cp_rd    : coverpoint get_gpr_name(ins.ops[0].val, ins.ops[0].key, "addi");
        cp_imm6  : coverpoint get_imm(ins.ops[2].val,"addi" ) {
            bins neg  = {[$:-1]};
            bins zero = {0};
            bins pos  = {[1:$]};
        }
    endgroup
```

### Steps to Reproduce
I have checked in the above code onto a `regression_test` branch of core-v-verif.  A run on Metrics shows that the RV32C coverage is now in place, but only the **c_add** and **c_addi** instructions are getting any coverage.  **c_addi16sp** and **c_addi4spn** are not hit (so we need to run more hits).   The other compressed cover groups are not hit because they are not implemented.

To run the example on `regression_test`, clone and run it in the usual way:
```
$ cd <my_work>
$ clone -b regression_test https://github.com/openhwgroup/core-v-verif regression_test
$ cd regression_test/cv32/sim/uvmt_cv32
$ make sanity SIMULATOR=<sim>
```
### Completion Criteria for this Issue
Implement cover_groups for the remaining RV32C instructions as per my example for the **c_add**, **c_addi**, **c_addi16sp** and **c_addi4spn** cover_groups.

strichmo, 2020-09-08:
The latest PR for the functional coverage should address that all compressed instructions from the ISS should be covered.

https://github.com/openhwgroup/core-v-verif/pull/222

The PR update now fires UVM warnings for any un-mapped instructions.  We will quickly move to make these fatals or errors.

MikeOpenHWGroup, 2020-10-08:
Thanks Steve, this can be closed.

closed 2020-10-08 by MikeOpenHWGroup