openhwfoundation/core-v-verif #165: License of Verification project is not correct

ShashankVM, 2020-07-17:
# Verification project cannot be licensed under Solderpad Hardware Licence
This Verification project cannot be licensed under Solderpad hardware Licence as it has a **dependency** on closed-source software, namely the Imperas Instruction Set Simulator. This is **not legal** according to Section 8 of https://opensource.guide/legal/.

### Type
* Other.

### Steps to Reproduce
Visit https://github.com/openhwgroup/core-v-verif/blob/master/LICENSE.md

### Additional context
Either remove the dependency on Imperas simulator or stop using the Solderpad Hardware Licence.

MikeOpenHWGroup, 2020-07-17:
With respect @ShashankVM, your interpretation of the situation is not correct.   Solderpad is a permissive open-source license which means that the project(s) licensed under it can use closed-source contributions.

closed 2020-07-17 by MikeOpenHWGroup

ShashankVM, 2020-07-17:
@MikeOpenHWGroup **permissive** open-source license means that **proprietary** **derived** works are allowed. That **does not mean that you can distribute proprietary code under an open-source licence, as your code does not *satisfy* the *definition* of open-source**.

MikeOpenHWGroup, 2020-07-17:
I am not a lawyer, but I disagree.   We can be dependent on close source code and we can distribute it (e.g. have compiled objects in our public repository).

rickoco, 2020-07-17:
Hello @ShashankVM - thanks for your comments, but to be clear, we are not distributing proprietary code and calling it open-source.  Our CORE-V Verification Test Bench leverages several commercial, closed source, tools with commercial System Verilog simulators being the most obvious and the Imperas ISS another example.  Within the OpenHW environment, we use dsim from Metrics, Xcelium from Cadence for SV simulation and the Imperas ISS to run our open-source test bench.  All of the stimulus in the test bench is open-source and available for any user to freely adopt / modify for use with whatever SV simulator or ISS reference model they would like to use.

ShashankVM, 2020-07-18:
I have to reply to Rick before I leave.
@rickoco SystemVerilog is an IEEE standard, not the Intellectual Property of a company. You cannot compare the use of standard SystemVerilog simulators to Imperas simulators because OVP, OVP APIs and the associated models are the intellectual property of Imperas. Your comparison of proprietary SystemVerilog simulators to Imperas simulators is like comparing the proprietary Intel C++ Compiler with MATLAB. C++ is a programming language that is an ISO standard; MATLAB is the intellectual property of MathWorks. Even if you use a proprietary C++ compiler in your open-source work that would not be a problem. But if you use MATLAB in your open-source work that would be meaningless, since you cannot use or **modify** that work without purchasing MATLAB. This issue is a lot like the [Java trap](https://www.gnu.org/philosophy/java-trap.en.html)

In conclusion, you cannot modify the golden reference model and use it without purchasing licence(s) for Imperas simulators.
Fact is the reference model will not be useful if it is not [technology agnostic](https://en.wikipedia.org/wiki/Reference_model).
Why should we not ask @zarubaf, the core creator of PULPino and director of engineering, OpenHW Group to comment?

ShashankVM, 2020-07-18:
Western Digital has also created open-source RISC-V cores written in SystemVerilog. Their Instruction Set Simulator is open-source, as it should be. https://github.com/westerndigitalcorporation/swerv-ISS

rickoco, 2020-07-18:
@ShashankVM - this will be my last reply on this topic as it is clear we'll need to 'agree to disagree'.  The OpenHW Group was established to curate well verified, high quality, open-source IP using the best tools available for the job (commercial or otherwise).  The CORE-V Verification Test Bench our community has built follows that strategy.  As I posted earlier, all of the stimulus in the test bench is open-source and available for any user to freely adopt / modify for use with whatever SV simulator or ISS reference model they would like to use. The user can disable the automatic inclusion of the Imperas ISS and use whatever ISS they choose.

ShashankVM, 2020-07-19:
Okay @rickoco

rickoco, 2020-07-19:
> Okay @rickoco. **I will create a separate fork of this entire project with an open-source ISS (or other open-source technology)**. My fork will also be licenced under the Solderpad Hardware Licence and I'll make sure it will be _vendor-neutral_.
> @MikeOpenHWGroup

Hello @ShashankVM that is exactly what we're expecting will occur with as users adopt the CORE-V Verification Test Bench which is how support for VCS and Questa was added beyond the original dsim and Xcelium work.  Feel free to stay engaged to be in sync with new capability commits in the test bench and to add back any improvements you make.

ShashankVM, 2020-07-21:
> > Okay @rickoco. **I will create a separate fork of this entire project with an open-source ISS (or other open-source technology)**. My fork will also be licenced under the Solderpad Hardware Licence and I'll make sure it will be _vendor-neutral_.
> > @MikeOpenHWGroup
> 
> Hello @ShashankVM that is exactly what we're expecting will occur as users adopt the CORE-V Verification Test Bench which is how support for VCS and Questa was added beyond the original dsim and Xcelium work. Feel free to stay engaged to keep your work in sync with new capability commits in the test bench and to add back any improvements you make.

Yes, @rickoco. Thank you.