openhwfoundation/core-v-verif #191: uvmt_cv32_step_compare.sv: MTVEC has mode Hardwired to 2'b01

eroom1966, 2020-08-14:
### Type
Indicate whether the type of problem you found:
* Functionally incorrect behavior

In the file https://github.com/openhwgroup/core-v-verif/blob/master/cv32/tb/uvmt_cv32/uvmt_cv32_step_compare.sv
this looks incorrect

```
             // MT: 2020-06-11
             //"miex"       : csr_val = `CV32E40P_CORE.cs_registers_i.miex_q;
             "mtvec"      : csr_val = {`CV32E40P_CORE.cs_registers_i.mtvec_q, 6'h0, 2'b01};
```

The mode bits are not hard wired 2'b01, they are initialized as 2'b01, but the LSB mtvec[0] is writable
this issue was exposed by a compliance test, I will prepare a pull request, I think it should be 
`             "mtvec"      : csr_val = {`CV32E40P_CORE.cs_registers_i.mtvec_q, 6'h0, `CV32E40P_CORE.cs_registers_i.mtvec_mode_q};`

Thx
Lee

strichmo, 2020-08-14:
@eroom1966 Good timing, I just saw the same failure in my testbench when integrating the ISS to the interrupt directed test.

I made the change above, however the ISS still believes that bit[0] (MTVEC.MODE) is still 1 after writing to MTVEC.

21823.000 ns            2118 000015e0 30579073 csrrw            x0, x15, 0x305      x15:00000300
# UVM_ERROR @ 21907.000 ns : uvmt_cv32_step_compare.sv(79) reporter [Step-and-Compare] mtvec expected=0x00000301 and actual=0x00000300 PC=0x00000fd8

Note that I am still using the older ISS model in core-v-verif.

Imperas, 2020-08-14:
@strichmo 
yes this also required a change to the reference model
so here is the strange thing, when I looked at the configuration of the envelope model for CV32E40P.
It was **explicitly** coded to have MODE hardwired as 'b01, this work was originally done by me and @GTumbush 
This was not done by oversight or accident, as this is **not** the default behavior of the base model
So I am wondering if the spec has evolved since we first implemented this, and we did indeed intend to have it behave this way ?

Either way, I have a release which I will hope to get a PR done today

MikeOpenHWGroup, 2020-08-14:
`git blame` clearly demonstrates that I was the one who added this in.  I've looked at the history of the RTL and cannot see where I do that idea from.  Clearly my bad.

strichmo, 2020-09-08:
The corev_rand_interrupt test should be randomizing mtvec at startup via riscv-dv.  I have added cover properties to the interrupt assertion set to ensure all interrupts are hit in vectored and fixed modes.  The directed test also covers both modes.

MikeOpenHWGroup, 2020-09-09:
Good work @strichmo.  Do you and @eroom1966 agree that this issue can be closed?   If so, @eroom1966 as the individual who opened the issue, you should be the one to close it.

MikeOpenHWGroup, 2020-10-19:
I have recommended this issue be closed and there has been no comment on this > 1 month, so I am closing.

closed 2020-10-19 by MikeOpenHWGroup