openhwfoundation/core-v-verif #219: Not all inputs to the DUT are driven

MikeOpenHWGroup, 2020-09-06:
This is a clone of cv32e40p issue [#485](https://github.com/openhwgroup/cv32e40p/issues/485).

@Silabs-ArjanB reported:
> This looks like an error of core-v-verif itself. Specifically not all inputs to cv32e40p_core are assigned at time 0 and some are not driven at at any time. All input signals need to be valid at all times, especially the clock input (which is not the case).

[image]

### Steps to Reproduce
```
$ git clone https://github.com/openhwgroup/core-v-verif
$ cd core-v-verif/cv32/sim/uvmt_cv32
$ make SIMULATOR=xrun sanity WAVES=1
```

### Additional context
I believe this bug has been latent in the testbench for some time.  It is now causing unique case violations at t=0 because of changes to the wave clock and reset are driven (see pr #214).

MikeOpenHWGroup, 2020-09-06:
Update.   I conducted a couple of experiments:

1. Drive `clk_i` to 'b0 starting at t=0.  This had no effect and the unique case violations were still reported.
2. Drive `clk_i` to 'b0 and `rst_ni` to 'b0 starting at t=0.  This clears up the unique case violations.

I believe (did not check) that the reason asserting reset at t=0 (rather than t>0) "fixes" the problem is because the debug and interrupt agents are reset by the same signal and both drive 'X until reset.

IMHO this is a simulation artifact that will probably not affect the real gates (after all, once reset happens, everything settles down).  Having said that, I think should look at fixing it in the RTL.

Silabs-ArjanB, 2020-09-07:
>> I think should look at fixing it in the RTL.

We do not consider this an RTL issue; it looks like a testbench issue.

Silabs-ArjanB, 2020-09-08:
@MikeOpenHWGroup FYI: The W,NOCOND issue should have been solved with https://github.com/openhwgroup/cv32e40p/pull/492. We do _not_ want (to mandate) assertion of reset at t=0

MikeOpenHWGroup, 2020-09-08:
Thanks @Silabs-ArjanB, I saw that.  Will close this issue as well.

closed 2020-09-08 by MikeOpenHWGroup