openhwfoundation/core-v-verif #221: ci_check does not correctly detect tool errors

nakengelhardt, 2020-09-07:
### Type
If a simulation run is aborted due to a tool error, not a simulation failure, the `ci_check` script can erroneously report `CI Check PASSED with known failure(s).`

### Steps to Reproduce
At e1b49e78, running `ci_check -s dsim` now sets CCOV=1 by default. If using an older version of dsim, the flags `-code-cov` and `-code-cov-scope-specs` are not supported, and the dsim runs are aborted with
```
dsim: error: invalid switch '-code-cov'
dsim: error: invalid switch '-code-cov-scope-specs'
Invoke 'dsim -help' for a list of valid options.
Copyright (C) 2017-2020 Metrics Technologies Inc. All rights reserved.
This program is licensed for the exclusive use of openhwgroup,
and may not be used otherwise without written permission from Metrics Technologies Inc.
/home/nak/master/cv32/sim/uvmt_cv32/dsim.mk:96: recipe for target 'comp' failed
make: *** [comp] Error 1
```

Despite these tests never running, at the end the script reports
```
CI Check PASSED with known failure(s).
OK to issue a pull-request.
```
This is due to the script counting the number of times that the strings `"SIMULATION PASSED"` and `"SIMULATION FAILED"` appear in the log files, but ignoring any logs that contain neither.

MikeOpenHWGroup, 2020-11-09:
Solved by @GTumbush in #345.

eroom1966, 2020-11-10:
@MikeOpenHWGroup @GTumbush 
Could I ask this to be re-opened ?
Greg has definitely moved this forward with his improvements.
When I now run ci_check I get the following

> ../cv32/sim/uvmt_cv32/xrun_results/hello-world/xrun-hello-world.log:                        SIMULATION PASSED
> ../cv32/sim/uvmt_cv32/xrun_results/illegal/xrun-illegal.log:                        SIMULATION PASSED
> ../cv32/sim/uvmt_cv32/xrun_results/debug_test_0/xrun-debug_test_0.log:                        SIMULATION PASSED
> ../cv32/sim/uvmt_cv32/xrun_results/csr_instructions/xrun-csr_instructions.log:                        SIMULATION PASSED
> ../cv32/sim/uvmt_cv32/xrun_results/riscv_arithmetic_basic_test_0/xrun-riscv_arithmetic_basic_test_0.log:                        SIMULATION PASSED
> CI Check FAILED: Expected 11 tests to run but found only 5 PASSED or FAILED messages

I know the exact reason for this, and it is something I have raised previously.
I think the make infrastructure is intending to be run from a CSH or TCSH ?
I use a BASH shell.
the smoking gun is the following output from make

> for (( idx=1; idx < $((1 + 1)); idx++ )); do \
> 	rm -f /home/moore/git/eroom1966/core-v-verif/cv32/sim/uvmt_cv32/xrun_results/corev-dv/corev_jump_stress_test/corev_jump_stress_test_$idx.S; \
> done
> **/bin/sh: 1: Syntax error: Bad for loop variable**

My understanding is that if you require make to use a specific SHELL interpreter when executing shell commands, then you MUST define what that SHELL should be - this is done by defining the variable SHELL in make
https://www.gnu.org/software/make/manual/html_node/Choosing-the-Shell.html

I have tried changing the Make infrastructure to use /bin/csh and /bin/tcsh by defining in the Common.mk - neither of which have proved successful

MikeOpenHWGroup, 2020-11-10:
@eroom1966, `ci_check` works for me on two platforms.  The IBM VM runs under Debian 10 and bash and the CMC VM is running CentOS and a cshell.

Can you provide more information?

eroom1966, 2020-11-10:
@MikeOpenHWGroup 
I think I realise what is happening

On a Ubuntu machine, the bourne shell is a symbolic link
/bin/sh -> dash

so if you run a /bin/sh - you **actually** get a dash shell
by Default make will use /bin/sh as its SHELL interpreter

so the commands in the makefiles on ubuntu are interpreted using dash, and in some cases as shown in my last post - are not correct syntax

good news - I have a fix, simply add the following to Common.mk

> export SHELL = /bin/bash

this overcomes the shell interpreter issues.
It still does not overcome my issue that ci_check reports 11 tests ran, but only 5 produces PASSED or FAILED

shall I make a PR to the Common.mk ?

MikeOpenHWGroup, 2020-11-10:
Great detective work @eroom1966.

> shall I make a PR to the Common.mk ?

Yes please!

GTumbush, 2020-11-10:
Hi @eroom1966 what is in the log files for the tests that ran but have no PASSED or FAILED

eroom1966, 2020-11-10:
PR done, can I ask one final thing
I attach my log from running ci_check - can you see from this, why I am not successfully getting all 11 tests running ?
I would be much happier creating PR's if I was satisfied that the ci_check was running as expected
[ci_check.log](https://github.com/openhwgroup/core-v-verif/files/5517695/ci_check.log)

MikeOpenHWGroup, 2020-11-10:
Hi @eroom1966.  None of the corev-dv tests are being run.   It looks like they are being generated, just not run.   Try this on the command line:
```
cd cv32/sim/uvmt_cv32
make corev-dv SIMULATOR=xrun
make gen_corev-dv test TEST=corev_jump_stress_test SIMULATOR=xrun
```
If that runs fails to run, `ls ../../tests/programs/corev-dv/corev_jump_stress_test` to see if you have an assembler program there.  It should have been created by the gen_corev-dv target.

eroom1966, 2020-11-10:
@MikeOpenHWGroup 

> make corev-dv SIMULATOR=xrun

I think this passes - but I get quite a few error messages including

> xmvlog: *W,NOTIND: unable to access -INCDIR /home/moore/git/eroom1966/core-v-verif/vendor_lib/google/riscv-dv/tests (No such file or directory).

and lots of messages of the form

> xmelab: *W,BNDWRN (/home/moore/git/eroom1966/core-v-verif/vendor_lib/google/riscv-dv/src/riscv_instr_cover_group.sv,647|29): Bit-select or part-select index out of declared bounds.
>   `FP_LOAD_INSTR_CG_BEGIN(fld, D)

when running 

> make gen_corev-dv test TEST=corev_jump_stress_test SIMULATOR=xrun

I get the following summary

> Number of demoted UVM_FATAL reports  :    0
> Number of demoted UVM_ERROR reports  :    0
> Number of demoted UVM_WARNING reports:    0
> Number of caught UVM_FATAL reports   :    0
> Number of caught UVM_ERROR reports   :    0
> Number of caught UVM_WARNING reports :    0
> 
> UVM_INFO @ 0 : uvm_report_server.svh(847) reporter [UVM/REPORT/SERVER] 
> --- UVM Report Summary ---
> 
> ** Report counts by severity
> UVM_INFO :  598
> UVM_WARNING :    0
> UVM_ERROR :    0
> UVM_FATAL :    0
> ** Report counts by id
> []     2
> [RNTST]     1
> [UVM/RELNOTES]     1
> [UVM/REPORT/CATCHER]     1
> [asm_gen]    29
> [callstack_gen]     6
> [cfg]     1
> [data_page_gen]     2
> [main]     8
> [pmp_cfg]     2
> [privil_seq]     1
> [riscv_instr]   516
> [sub_1]     5
> [sub_2]     5
> [sub_3]     5
> [sub_4]     4
> [sub_5]     4
> [uvm_test_top]     5
> 
> Simulation complete via $finish(1) at time 0 FS + 231
> /home/nda/cadence/installs/XCELIUM2003/tools/methodology/UVM/CDNS-1.2-ML/sv/src/base/uvm_root.svh:551     $finish;
> xcelium> exit
> TOOL:	xrun(64)	20.03-s010: Exiting on Nov 10, 2020 at 14:14:19 GMT  (total: 00:00:01)
> make: *** [/home/moore/git/eroom1966/core-v-verif/cv32/sim/uvmt_cv32/xrun.mk:334: gen_corev-dv] Error 1

finally in the directory  ../../tests/programs/corev-dv/corev_jump_stress_test I only see

> corev-dv.yaml  test.yaml

Thx

eroom1966, 2020-11-10:
@MikeOpenHWGroup @GTumbush 
OK, the problem for me is that the call to xrun is returning a non-zero code which is not trapped by the caller
so this command seems to pass but returns the value 1

>         $(XRUN) -R $(XRUN_RUN_FLAGS) \
>                 -xceligen rand_struct \
>                 -l $(TEST)_$(GEN_START_INDEX)_$(GEN_NUM_TESTS).log \
>                 +start_idx=$(GEN_START_INDEX) \
>                 +num_of_tests=$(GEN_NUM_TESTS) \
>                 +UVM_TESTNAME=$(GEN_UVM_TEST) \
>                 +asm_file_name_opts=$(TEST) \
>                 $(GEN_PLUSARGS)

I worked around by ignoring he return code by prefixing with the '-' at the beginning of the command
but the real question is why is this command returning a non-zero (failure) code

eroom1966, 2020-11-10:
OK, I think I see what is happening
ubuntu is not _officially supported_ by Cadence, this is causing a non zero return code to be generated

With my change above my ci_check passes

MikeOpenHWGroup, 2020-11-10:
Whew!   Glad to see you grounded this out @eroom1966.  I could see what was happening, but could not understand why.  I did see that "unsupported OS" warning in your logfile and wondered about it, but said to myself, "nah, that _can't_ be it".

Can we close this one?

eroom1966, 2020-11-10:
yes please close

closed 2020-11-10 by MikeOpenHWGroup