openhwfoundation/core-v-verif #227: Tool-specific Makefile for Aldec Riviera-PRO

MikeOpenHWGroup, 2020-09-10:
Hi @dawidzim, you've been making useful recommendations and pull-request to this repository so I'd like to see if you can do something more.  As you are probably aware, core-v-verif currently supports Metrics **dsim**, Cadence **Xcelium**, Mentor's **Questa** and Synopsys' **VCS** SystemVerilog simulators.   We also support Verilator in our non-UVM "core" testbench.   It would be great to add support for Aldec **Riviera-PRO**.  Are you in a position to help out with that?

### Completion Criteria
1. Addition of `rpro.mk` to `core-v-verif/cv32/sim/uvmt_cv32` directory.
2. Ability to run individual testcases.
3. Integration into core-v-verif test specification methodology.

### Additional context
Ideally, the stability of the Makefile updates would allow an member of the open source community that has access to Aldec Riviera-PRO to use core-v-verif in the same manner as members who use other SystemVerilog simulators.

danielmlynek, 2020-09-11:
We are working on supporting core-v-verif in Riviera-PRO. All those issues reported by Dawid was found during that process.
Closing all those reported issue/pull request will help us to achieve core-v-verif support in our tool.

Could you elaborete point 3 "Integration into core-v-verif test specification methodology."

We also want to test core-v-verif cosimulation with Imperas model. At the moment we asked for license and waiting on responce. Can get this license as openhwgroup collaborators?

MikeOpenHWGroup, 2020-09-11:
> We are working on supporting core-v-verif in Riviera-PRO.

In this context, who is "we"?  In any case, I am pleased to see a Riviera-PRO user looking at this project and will work to help you get going.

> Could you elaborete point 3 "Integration into core-v-verif test specification methodology."

We have just introduced a yaml based format for describing testcases.  We use this to generate Makefile rules that support a unified command-line for running testcases, (almost) independent of the simulator they are using.  Currently this does not support Riviera-PRO because up to now we have not had an users of that simulator (well, none that we are aware of).  It would be a significant contribution to the community if you would provide the necessary updates to our scriptware and Makefiles to support Rivieria-PRO.

> We also want to test core-v-verif cosimulation with Imperas model.

To obtain the commercial license you'll either need to purchase it from Imperas or become a member of OpenHW.  If you are interested in OpenHW membership, please send me an email at mike@openhwgroup.org and we can get the ball rolling on that.

danielmlynek, 2020-09-14:
> In this context, who is "we"? In any case, I am pleased to see a Riviera-PRO user looking at this project and will work to help you get going.

We at Aldec.

Thank you for information. 
We will continue our work to close this issue.

MikeOpenHWGroup, 2020-10-16:
Complete as of pr #269.  Thanks @danielmlynek!

closed 2020-10-16 by MikeOpenHWGroup