openhwfoundation/core-v-verif #239: Very basic performance counter test

Silabs-ArjanB, 2020-09-15:
Here is a very limited directed test that was used for debugging https://github.com/openhwgroup/cv32e40p/issues/429. Please use it as you see fit for verification of the performance counters.

[hpmcounter_test.txt](https://github.com/openhwgroup/core-v-verif/files/5224901/hpmcounter_test.txt)

eroom1966, 2020-09-17:
@Silabs-ArjanB @MikeOpenHWGroup 
I am working with @silabs-oysteink in order to incorporate the hpmcounters into the step/compare
we have initially seen an issue with intra instruction changes to mcycle - this may also affect other counters, so we need to devise a way to handle this efficiently

Let me try to explain the issue.
The RM always lags the RTL by one instruction retirement, the flow looks like this (pseudo-code)
while (1) {
  step_rtl_to_retire()
  step_ovp_to_retire()
  compare_state()
}

in reality, the step_rtl_to_retire() takes N cycles, depending upon the instruction type and whether any exceptions (sync/async) are raised.
Because it takes N cycles the value of mcycle will be modified N times, and we need to know the value of the mcycle which was visible at the point of instruction execution, for example, if we had the instruction
csrr t1, mcycle
the value written to t1 will not be the value of mcycle when the instruction retires, mcycle will have continued incrementing.

We need to inform the RM of the values of these realtime counters as they were 'architecturally visible' at the time of instruction execution - this will allow us to remain in sync

In conclusion, we cannot simply read the values of hpmcounter* at the point of instruction retire, it seems we need the values at the point of instruction execute.

all help gratefully received.

Silabs-ArjanB, 2020-09-17:
@eroom1966 FYI (not sure if this info will help you): The minstret counter in the RTL increments on 'issue' of an instruction instead of 'retirement'; this is easier to implement as only one instruction will be issued at a time whereas multiple instructions can retire at the same time in the RTL. The difference should however not be noticeable by a program executing on the RTL (as every issued instruction will eventually retire). Of course you are poking directly into some RTL registers, so for you this effect could be visible. Also please note that some fixes have been made but not yet merged: https://github.com/openhwgroup/cv32e40p/pull/504 (but these do not really impact what you are stating above)

Silabs-ArjanB, 2020-09-17:
@eroom1966 In my opinion a cycle by cycle comparison is maybe not the way to go. It would be much easier to execute random programs and then compare the result of a counter read via a regular CSR instruction (as opposed to poking in the RTL).

eroom1966, 2020-09-17:
> 
> 
> @eroom1966 In my opinion a cycle by cycle comparison is maybe not the way to go. It would be much easier to execute random programs and then compare the result of a counter read via a regular CSR instruction (as opposed to poking in the RTL).

@Silabs-ArjanB it would be easier for sure, but not definitive. For example we would not have identified some of the issues such as the dpc/dcsr issues recently.
This issue is only related to counters that change between an instruction using a value, and then retireing as the value changes, mcycle being a very good example.
mcycle cannot be calculated in an instruction accurate model as it only models the architectural state.
In which case to ensure an instruction such as 
`csrr t1, mcycle`
executes correctly in the architectural model, we must tell it what value to use for mcycle, when executing this instruction
otherwise the value of t1 will propagate throughout the entire machine state

minstret is not an issue, and will work correctly for the reasons you outline

This problem only exists for counters changing state after execution, but before retirement.
In this case we MUST capture the value of the counter at the execution stage to ensure correct propagation

In terms of comparing the performance counters themselves, I do not think that is possible or even useful, but we can and must compare the effect of those counters in the rest of the machine state as they propagate through csrr instruction execution.
Thx
Lee

MikeOpenHWGroup, 2020-09-17:
@eroom1966, @Silabs-ArjanB, @silabs-oysteink, @strichmo 

The topic of this issue has changed so I have edited the subject.  Let's try to maintain single-topic issues in GitHub.

Sooner or later, this issue pops up in every verification project.  To avoid us getting struck in cycle of ever-diminishing returns, I am inclined to agree with Arjan.  However, Lee's example (`csrr t1 mcycle`) illustrates that we cannot simply disable step-and-compare of mcycle until the end of simulation because the check on t1 will fail.  So we need to consider all CSRs subject to "intra-instruction changes" and articulate a verification strategy for each one. This strategy can be coded into the `compare_state()` method from Lee's pseudo-code at  the start of this issue.

The CSRs discussed in this thread are `mcycle` and `minstret`.   I'll suggest two strategies that can be used for these CSRs.  Note that both assume that the `compare_state()` method is aware of the retired instruction (and I am not sure this is the case).

**Strategy 1: ignore them**. Simply do not compare `mcycle` and `minstret` CSRs, including and destination GPRs written to with the value of these ignored CSRs.  (Of course, mcycle and minstret should be checked at end-of-sim.)

**Startegy 2: accept uncertainty**.  Correct me if I am wrong, but I do not believe the exact value of these counters is important until the end of the simulation.  For both of these CSRs (and any destination GPRs), the compare_state() method should accept and value in the range of predicted_value +/- predicted_uncertainty.   For mcycle the predicted uncertainty is likely 10s of cycles (longest instruction plus pipeline depth), which the uncertainty of minstret is likely only one.

Does this work?  Are there any other CSRs subject to "intra-instruction changes""

eroom1966, 2020-09-17:
I have been investigating this issue (all day!) and I think I have a solution
Firstly I don't think we can directly compare CSR's whose values are not architecturally deterministic
Its simply not possible, mcycle is a great example of this.
Fortunately, it is still possible to compare their propagation effect to GPR's - IF we can inject the CSR correct values prior to the execution of instructions which rely upon access to registers such as mcycle, minstret (and others)

The goal is to inject the performance counter CSR values into the RM, immediately prior to using them as part of an assignment to a GPR, this insures that the RTL and RM remain in sync with respect to data propagation.

I am confident I have worked out how to do this, and I have submitted a PR for @silabs-oysteink to try. My results are that the RTL and RM remain in-sync

If this is the case, I think this is a general strategy for ensuring access to these performance counters are valid for matching the execution in both the RTL and RM

Lets wait for an update from @silabs-oysteink

silabs-oysteink, 2020-09-17:
I can confirm the the latest changes are good, no miscompares on mcycle and minstret. Issued a PR to core-v-verif/master with all updates.

Silabs-ArjanB, 2020-09-17:
Hi @eroom1966 Can you confirm that you only need to 'inject CSR correct values' for 'cycle count' and not for any of the other counted events?

eroom1966, 2020-09-17:
@Silabs-ArjanB 
I am not sure what other counted events will be available, I am guessing that ALL of the hpmcounter values will need passing as they are potentially deeply embedded with the hardware ?

It is difficult to know, but if it is recording things such as cache hit/miss, pipeline stalls ....
then these will need sharing with the RM

Silabs-ArjanB, 2020-09-18:
Hi @eroom1966 You can find the performance counter events here: https://core-v-docs-verif-strat.readthedocs.io/projects/cv32e40p_um/en/latest/perf_counters.html. For the following I would expect you will _not_ need 'value passing' to get them verified: INSTR, LD, ST, JUMP, BRANCH, BRANCH_TAKEN, COMP_INSTR.

eroom1966, 2020-09-18:
@Silabs-ArjanB 
Hi Arjan,
I would like to run this test, is it checked in, or do I need to copy the attachment into my own fork
Thx
Lee

Silabs-ArjanB, 2020-09-18:
Hi @eroom1966 Maybe the NUM_MHPMCOUNTERS  parameter is configured differently between RTL and ISS? I just gave that C/assembly code as an example; I do not know where this should be checked in inside core-v-verif

eroom1966, 2020-09-18:
@Silabs-ArjanB 
that is a very good point @MikeOpenHWGroup is there an intended default setting for NUM_MHPMCOUNTERS, and does this need to be a parameter on the RM to replicate this behavior ?

Silabs-ArjanB, 2020-09-18:
@eroom1966 There is an ongoing discussion about the default param settings, but for NUM_MHPMCOUNTERS it will be 1 for sure.

eroom1966, 2020-09-18:
@Silabs-ArjanB 
If I understand correctly, this parameter simply acts as a bit mask to the inhibit register ?

Silabs-ArjanB, 2020-09-18:
Hi @eroom1966 Yes, roughly. The parameter is described at https://core-v-docs-verif-strat.readthedocs.io/projects/cv32e40p_um/en/latest/perf_counters.html. If NUM_MHPMCOUNTERS=1, then hpmcounter3(h) and related inhibit bits are writeable, if NUM_MHPMCOUNTERS=2, then hpmcounter3(h), hpmcounter4(h) and related inhibit bits are writeable, etc.

Silabs-ArjanB, 2020-09-23:
Made fixes in test case to make it more tolerant against compilation options

[hpmcounter_test.txt](https://github.com/openhwgroup/core-v-verif/files/5267310/hpmcounter_test.txt)

MikeOpenHWGroup, 2020-10-19:
Thanks @Silabs-ArjanB.  Your test will be part of the baseline regressions as soon as #292 is merged in.

MikeOpenHWGroup, 2020-10-26:
Hi @Silabs-ArjanB, #292 has been merged, so I would like to close this issue.   The test is failing, and that is being tracked in #304.  Please let me know if you agree.

Silabs-ArjanB, 2020-10-26:
Hi @MikeOpenHWGroup , yes sure, I will close this one.

closed 2020-10-26 by Silabs-ArjanB