openhwfoundation/core-v-verif #242: Please add license info to comment headers

MikeOpenHWGroup, 2020-09-16:
Hello @nakengelhardt, I have a request.  As part of the OpenHW Group's corporate goverance we are being audited by the [Eclipse Foundation](https://www.eclipse.org/org/foundation/) to ensure our open-source contributions are, in fact, legally open course artifacts.  Part of this audit includes having the proper license headers and Eclipse's scriptware has flagged several of your contributuons.   Could you prepend the following (or equivalent) to your contributions in `cv32/mcy`?  You can either replace <Contributor's Organization> with your company's legal name or delete the line altogether (leaving the OpenHW Group line).

The "SPDX-License-Identifier" (last line) is the most important.

Of course, some of your files will need other comment delimiters, but I am sure you get the idea.   Thanks!

//
// Copyright 2020 OpenHW Group
// Copyright 2020 <Contributor's Organization>
//
// Licensed under the Solderpad Hardware License, Version 2.0 (the "License");
// you may not use this file except in compliance with the License.
// You may obtain a copy of the License at
//
//     https://solderpad.org/licenses/
//
// Unless required by applicable law or agreed to in writing, software
// distributed under the License is distributed on an "AS IS" BASIS,
// WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
// See the License for the specific language governing permissions and
// limitations under the License.
// 
// SPDX-License-Identifier: Apache-2.0 WITH SHL-2.0
//

nakengelhardt, 2020-09-16:
The tool configuration files (.sby, .mcy) don't allow for comments so I can't insert license information into them directly. How should I handle that?

MikeOpenHWGroup, 2020-09-16:
OK, that is fine.  We have a few other examples of this as well, for example `.metrics.json`.   Hopefully there are not too many of these and I'll just get individual waivers for them.

PR #243 (2020-09-16): mcy: add license headers 
Fixes #242.

`config.mcy` and `test_eq.sby` don't support comments and therefore cannot have license headers included.
closed 2020-09-16 by MikeOpenHWGroup