openhwfoundation/core-v-verif #253: ci_check fails

Silabs-ArjanB, 2020-09-23:
In core-v-verif (hash 891805fe096181421896b640450d72a771744d98) the ci_check fails on xrun-illegal and xrun-hello-world. There are also various 'compliance test failures' visible but they do not end up in below summary:

TOOL:	xrun(64)	20.06-a001: Exiting on Sep 23, 2020 at 05:26:29 CDT  (total: 00:00:08)
../cv32/sim/uvmt_cv32/xrun_results/illegal/xrun-illegal.log:                   **SIMULATION FAILED - ABORTED**
../cv32/sim/uvmt_cv32/xrun_results/hello-world/xrun-hello-world.log:                   **SIMULATION FAILED - ABORTED**
../cv32/sim/uvmt_cv32/xrun_results/corev_jump_stress_test_1/xrun-corev_jump_stress_test_1.log:                        SIMULATION PASSED
../cv32/sim/uvmt_cv32/xrun_results/corev_arithmetic_base_test_0/xrun-corev_arithmetic_base_test_0.log:                        SIMULATION PASSED
../cv32/sim/uvmt_cv32/xrun_results/corev_jump_stress_test_0/xrun-corev_jump_stress_test_0.log:                        SIMULATION PASSED
../cv32/sim/uvmt_cv32/xrun_results/corev_arithmetic_base_test_1/xrun-corev_arithmetic_base_test_1.log:                        SIMULATION PASSED
../cv32/sim/uvmt_cv32/xrun_results/corev_rand_instr_test_0/xrun-corev_rand_instr_test_0.log:                        SIMULATION PASSED
../cv32/sim/uvmt_cv32/xrun_results/riscv_ebreak_test_0/xrun-riscv_ebreak_test_0.log:                        SIMULATION PASSED
../cv32/sim/uvmt_cv32/xrun_results/csr_instructions/xrun-csr_instructions.log:                        SIMULATION PASSED
../cv32/sim/uvmt_cv32/xrun_results/corev_rand_instr_test_1/xrun-corev_rand_instr_test_1.log:                        SIMULATION PASSED
../cv32/sim/uvmt_cv32/xrun_results/riscv_arithmetic_basic_test_0/xrun-riscv_arithmetic_basic_test_0.log:                        SIMULATION PASSED

CI Check FAILED with unknown failures.

MikeOpenHWGroup, 2020-09-23:
Investigating.  Thanks for the notification.

Silabs-ArjanB, 2020-09-23:
Hi @MikeOpenHWGroup I already received a note that makes me think it might have to do with the xcelium version I used, so I think you can ignore this issue for now until I have tried the recommended xcelium version.

Silabs-ArjanB, 2020-09-23:
Hi @MikeOpenHWGroup , the 'illegal' and 'hello-world' test pass when moving to an older xcelium version (Steve is aware of the details). The Compliance Test signature checks still fail, but it does not lead to the ci_check to fail. For example the following is part of my output:

```
make[3]: Leaving directory `/home/arbink/openhw/verif/core-v-verif/vendor_lib/riscv/riscv-compliance/riscv-test-suite/rv32imc'
riscv-test-env/verify.sh

Compare to reference files ... 

Check               C-ADDI16SP ... OK
Check               C-ADDI4SPN ... OK
Check                   C-ADDI ... OK
Check                    C-ADD1,28d0
< 00000000
< 00000001
< ffffffff
< 00007fff
< 00008000
< 00000001
< 00000002
< 00000000
< 00008000
< 00008001
< ffffffff
< 00000000
< fffffffe
< 00007ffe
< 00007fff
< 00007fff
< 00008000
< 00007ffe
< 0000fffe
< 0000ffff
< 00008000
< 00008001
< 00007fff
< 0000ffff
< 00010000
< 00000000
< 00000000
< 00000000
 ... FAIL
```

MikeOpenHWGroup, 2020-09-23:
This is a known issue with the Compliance suite that is not part of the cv32e40p simulation.  The test-program that runs on the cv32e40p, and is step-and-compared against the ISS, does not fail (which is why `ci_check` does not report a failure).

As you may recall, the strategy for using the RISC-V Compliance suite is to use it "out of the box" without modification.  This means we are using the Compliance suite's makefile to compile the suite and these always want to run the compiled code against OVPsim.  In that step, there is a signature compare issue.  This is essentially a don't-care for us.

Nevertheless, I want to clean things up and Lee and I will be getting it once the debug and interrupt issues between the ISS and RTL are all resolved (we are close, I think).

Silabs-ArjanB, 2020-09-23:
Hi @MikeOpenHWGroup Okay, thank you for the explanation. I will change the title of this issue to 'signature check mismatch' or something like that.

Silabs-ArjanB, 2020-09-24:
Hi @MikeOpenHWGroup Could you please also give a command line for how to run the Compliance suite stand-alone (i.e. not as part of ci_regress). (I am using make cv32-riscv-compliance-tests, but that one (still) fails on some out-of-bounds read)

MikeOpenHWGroup, 2020-10-26:
Will do.   Please track #302.

MikeOpenHWGroup, 2020-11-23:
Issue resolved as soon as pr #401 is merged in.

MikeOpenHWGroup, 2020-11-24:
Hi @Silabs-ArjanB, this issue is resolved.  Please have a look and close this issue if you agree.

Silabs-ArjanB, 2020-11-24:
Hi @MikeOpenHWGroup Are you sure that the signature is checked for the RTL sim? I don't see the '... OK' that I would expect.

MikeOpenHWGroup, 2020-11-24:
The signatures are dumped by the program and both the RTL and ISS produce signature outputs to stdout.  The UVM environment does not have a mechanism to diff these and there are no plans to add that capability.

Silabs-ArjanB, 2020-11-24:
Hi, I would like to request this then as a future improvement (not gating RTL freeze) as this is a valuable test for RTL designers to run without the availability of an ISS.

MikeOpenHWGroup, 2020-11-30:
Hi @Silabs-ArjanB .  With the merging in of #406, the Compliance regression now also dumps and compares signatures.  See the README in the cv32/sim/uvmt_cv32 directory for the required commands.  Note that this is currently only working with Xcelium and dsim.  Steve will add Questa support and I will reach out to other Members to add VCS and Riviera-PRO support.

Silabs-ArjanB, 2020-11-30:
Hi @MikeOpenHWGroup Thank you very much for adding this.

closed 2020-11-30 by Silabs-ArjanB