openhwfoundation/core-v-verif #256: Assertion firing: Exception not entered correctly after ebreak with dcsr.ebreak=0

MikeOpenHWGroup, 2020-09-25:
There is a static test, `riscv_ebreak_test_0` that has been in regression for a couple of months now.  It was originally generated by riscv-dv but has been manually modified to work with the core-v-verif environment and is now considered a "directed test".  After merging in the latest [debug coverage and assertions](https://github.com/openhwgroup/core-v-verif/pull/249), the test is failing due to the follow assertion firing:
```
UVM_ERROR @ 9358.800 ns : uvmt_cv32e40p_debug_assert.sv(102) reporter [CV32E40P_DEBUG_ASSERT] Exception not entered correctly after ebreak with dcsr.ebreak=0
```

Even though this test has been passing for a couple of months now, it is entirely possible that the test was not sensitive to the bug flagged by the assertion, in which case the new assertion has found an RTL bug.  it is equally possible that the testcase and/or assertion itself has a bug.

To reproduce:
```
git clone https://github.com/openhwgroup/core-v-verif
cd core-v-verif/cv32/sim/uvmt_cv32
make test SIMULATOR=<sim> TEST=riscv_ebreak_test_0
```

silabs-oysteink, 2020-09-25:
This looks very much like an assertion bug. I used the wrong source for mtvec when hooking up the interface to the assertions.  I there are also a problem with the delay between the ebreak instruction and when the assertion is expecting the core to jump to the exception. I will debug and submit a PR when it's fixed.

MikeOpenHWGroup, 2020-09-28:
Test now passes as of core-v-verif hash fac0f25.

closed 2020-09-28 by MikeOpenHWGroup