openhwfoundation/core-v-verif #263: Assertion failures in riscv_ebreak_test: test bench reports passed

aimeepsutton, 2020-09-29:
## Bug Title
Assertion failures observed when running riscv_ebreak_test with USE_ISS=NO. Despite the errors, the test bench reports PASSED at the end of the simulation.

### Type
* Functionally incorrect behaviour
* Confusing or extraneous status or error messages

### Steps to Reproduce
Please provide:
1. URL to branch that exhibits the issue.
`git@github.com:openhwgroup/core-v-verif.git (master) 0b1006d14764dc82a720dd253b7f4647e3201069`

2. Command line  
`cd cv32/sim/uvmt_cv32; make test TEST=riscv_ebreak_test SIMULATOR=dsim USE_ISS=NO`

3. Logfile and/or wave-dump info (screen shots can be useful)
```
UVM_INFO @ 114.300 ns : uvmt_cv32_firmware_test.sv(148) uvm_test_top [TEST] Started RUN
=E:[AssertFailed] Assertion started at t=59964300 FAILED at t=59964300: /home2/asutton/repos/openhw/core-v-verif/core-v-cores/cv32e40p/rtl/cv32e40p_mult.sv:348
=E:[AssertFailed] Assertion started at t=60009300 FAILED at t=60009300: /home2/asutton/repos/openhw/core-v-verif/core-v-cores/cv32e40p/rtl/cv32e40p_mult.sv:348
=E:[AssertFailed] Assertion started at t=60054300 FAILED at t=60054300: /home2/asutton/repos/openhw/core-v-verif/core-v-cores/cv32e40p/rtl/cv32e40p_mult.sv:348
=E:[AssertFailed] Assertion started at t=60099300 FAILED at t=60099300: /home2/asutton/repos/openhw/core-v-verif/core-v-cores/cv32e40p/rtl/cv32e40p_mult.sv:348
=E:[AssertFailed] Assertion started at t=60144300 FAILED at t=60144300: /home2/asutton/repos/openhw/core-v-verif/core-v-cores/cv32e40p/rtl/cv32e40p_mult.sv:348
UVM_INFO @ 125700.300 ns : uvmt_cv32_firmware_test.sv(157) uvm_test_top [TEST] Finished RUN: exit status is 0
UVM_INFO @ 125700.300 ns : uvm_objection.svh(1270) reporter [TEST_DONE] 'run' phase is ready to proceed to the 'extract' phase
UVM_INFO @ 125700.300 ns : uvmt_cv32_base_test.sv(328) uvm_test_top [END_OF_TEST] DUT WRAPPER virtual peripheral signaled exit_value=0.
UVM_INFO @ 125700.300 ns : uvm_report_server.svh(847) reporter [UVM/REPORT/SERVER] 
--- UVM Report Summary ---

Quit count :     0 of     5
** Report counts by severity
UVM_INFO :   25
UVM_WARNING :    0
UVM_ERROR :    0
UVM_FATAL :    0
** Report counts by id
[BASE TEST]     5
[CLKNRST]     3
[CORE_CNTRL_IF]     1
[DEBUGCOVG]     1
[DUT_WRAP]     2
[END_OF_TEST]     1
[INTERRUPTCOVG]     1
[RNTST]     1
[RST_VSEQ]     3
[TEST]     3
[TEST_CFG]     1
[TEST_DONE]     1
[UVM/RELNOTES]     1
[rv32isa_covg]     1

SVA Summary:  82 assertions, 3434118 evaluations, 718378 nonvacuous passes, 32 disables, 5 failures
SVA Summary:  190 cover property statements, 7959860 evaluations, 0 nonvacuous passes, 190 statements not covered

uvmt_cv32_tb.end_of_test: *** Test Summary ***

    PPPPPPP    AAAAAA    SSSSSS    SSSSSS   EEEEEEEE  DDDDDDD     
    PP    PP  AA    AA  SS    SS  SS    SS  EE        DD    DD    
    PP    PP  AA    AA  SS        SS        EE        DD    DD    
    PPPPPPP   AAAAAAAA   SSSSSS    SSSSSS   EEEEE     DD    DD    
    PP        AA    AA        SS        SS  EE        DD    DD    
    PP        AA    AA  SS    SS  SS    SS  EE        DD    DD    
    PP        AA    AA   SSSSSS    SSSSSS   EEEEEEEE  DDDDDDD     
    ----------------------------------------------------------
                        SIMULATION PASSED                     
    ----------------------------------------------------------
```

### Additional context
None

Silabs-ArjanB, 2020-09-29:
Hi, this might be caused by X propagation via the data_* interface (we saw the same behavior yesterday)

MikeOpenHWGroup, 2020-09-29:
I was not able to reproduce this issue unless I added `USE_ISS=NO` (as specified by @aimeepsutton).  Its odd that the Assertions do not fire when the ISS is running (omit USE_ISS on the make command line).   Investigating.

MikeOpenHWGroup, 2020-09-30:
I do not fully understand why yet, but in the case without the ISS, there is 'bX on rdata for one cycle.  Eventually this makes its way to the assertion which fires.    When running _with_ the ISS, there are no 'bX on the bus and the test-program executes as you'd expect.

When the ISS is used the step-and-compare gaps the core's clock, so that is where I will start looking tomorrow.

MikeOpenHWGroup, 2020-10-01:
This is a testbench issue (well, not really, please read on).

When the ISS is used, the testbench will replace all unused instruction and data memory with random data.  This is necessary because the ISS maps 'bX to 'b0 and this can cause step-and-compare mismatches.  When the ISS is _not_ used, this initialization is skipped, leaving 'bX in some RTL instruction and/or data memory locations.  In this particular test the core fetched uninitialized memory and this caused Xs to flow through the pipeline and eventually get picked up by an assertion.

This is resolved in pr #267 by always initializing the RTL's I&D memory, whether or not the ISS is used.

aimeepsutton, 2020-10-01:
Sounds good Mike. What about the related issue that the test reports passed despite assertion failures?

MikeOpenHWGroup, 2020-10-01:
Sigh.   I've got a student in Italy working on that and they are struggling.  The template is all there and I could do it in a day, but I simply have too many other higher priority items on my plate.

PR #267 (2020-10-07): Initialize RTL instruction memory even when ISS not used
Hi Greg.  You've probably forgotten, but back in May of this year you updated the `dut_wrap` module to initialize unused instruction memory which could lead to step-and-compare mismatches because the RTL and TB model 4-state values and the ISS models 2-state values.

In this PR I have slightly modified this code so that the RTL's instruction memory is _always_ initialized, even if the ISS is not used.  This resolves #263.
closed 2020-10-07 by strichmo