openhwfoundation/core-v-verif #268: CSR step and compare mistiming with random stalls

strichmo, 2020-10-01:
### Type
Step and compare fault in the presence of random stalls.

When randomly stalling the CV32E40P OBI interfaces, the CSR timing updates can becoming unpredictable with respect to the 
instruction retirement events.  Currently the step and compare module will compare CSRs using snoops from the design to the ISS state at each instruction retirement.  However in the presence of stalls, the CSR can be updated before the previous instruction retires, creating a miscompare error.

### Steps to Reproduce
1. Clone this branch: https://github.com/openhwgroup/core-v-verif.git
2. Note that this branch also points to a cv32e40p from one of my branches since a new tracer is required.  The branch above points to my branch.
3. You must comment out the line in uvmt_cv32_step_compare.sv
-           ignore = 1; // FIXME:strichmo:THIS MUSDT BE FIXED:Need to resolve stalled pipeline issues with CSR matching
+           //ignore = 1; // FIXME:strichmo:THIS MUSDT BE FIXED:Need to resolve stalled pipeline issues with CSR matching
4. Run this:
% makecv32 comp_corev-dv gen_corev-dv test TEST=corev_rand_instr_stall SEED=10

There should be failures as such:
UVM_ERROR @ 4536.300 ns : uvmt_cv32_step_compare.sv(82) reporter [Step-and-Compare] mscratch expected=0x0000000f and actual=0x00000004 PC=0x000003da
UVM_ERROR @ 11520.300 ns : uvmt_cv32_step_compare.sv(82) reporter [Step-and-Compare] mscratch expected=0x0000000d and actual=0x00000001 PC=0x000007d8

The following is an example of the failure:

MikeOpenHWGroup, 2020-10-01:
Hi @eroom1966, I assigned this to you, not because I think this is "your" problem, but just to ensure you received an email notification that the issue was created.

eroom1966, 2020-10-02:
@MikeOpenHWGroup I will schedule some time to take a look thx

eroom1966, 2020-10-05:
@MikeOpenHWGroup received and acknowledged.
I am not sure how I can help out here
I think Steve has nailed this one accurately.
We need a solid method of extracting the CSR values for comparison to the RM -OR- a method to know when the sampling points are valid

MikeOpenHWGroup, 2020-11-04:
As per discussion with other Committers today, this issue does not gate RTL Freeze of CV32E40P so it is waived for now.   (Note that a new method for waiving core-v-verif issues is under investigation.)

eroom1966, 2022-11-04:
Is this still open
This should go away I hope once the move to ImperasDV is complete ?

MikeOpenHWGroup, 2022-11-04:
Agreed.

eroom1966, 2022-11-04:
Unfortunately I cannot close the issue

eroom1966, 2023-02-21:
I cannot close this issue - removing my assignment