openhwfoundation/core-v-verif #302: Compliance test still failing - cv32-riscv-compliance-tests

Silabs-ArjanB, 2020-10-23:
(I know this has been reported numerous times before, but there seems to be the impressions that this test is passing).

The compliance test is still failing.

commit 4c179c86d1ecdca9467cc670d67b3d44d2ff7e51
cd cv32/sim/uvmt_cv32
make cv32-riscv-compliance-tests

This runs for a while (using xrun, corev-openhw-gcc-centos7-20200913, USE_ISS=YES) and then produces:

I_LB_01..# Test Begin Reserved regs ra(x1) a0(x10) t0(x5)
# Test part A1 - test base address + 0
# Test part A1  - Complete
# Test part A2 - test base address - 1
# Test part A2  - Complete
# Test part A3 - test base address + 1
# Test part A3  - Complete
# Test part A4 - test base address - 2048
out of bounds read from fffff9e8
xmsim: *F,FATSEV (/home/arbink/openhw/verif/core-v-verif/cv32/tb/core/mm_ram.sv,414): (time 1050778800 PS).
uvmt_cv32_tb.dut_wrap.ram_i.read_mux
xmsim: *W,SLFINV: Call to process::self() from invalid process; returning null.

uvmt_cv32_tb.end_of_test: *** Test Summary ***

    FFFFFFFF   AAAAAA   IIIIII  LL        EEEEEEEE  DDDDDDD       
    FF        AA    AA    II    LL        EE        DD    DD      
    FF        AA    AA    II    LL        EE        DD    DD      
    FFFFF     AAAAAAAA    II    LL        EEEEE     DD    DD      
    FF        AA    AA    II    LL        EE        DD    DD      
    FF        AA    AA    II    LL        EE        DD    DD      
    FF        AA    AA  IIIIII  LLLLLLLL  EEEEEEEE  DDDDDDD       
    --------------------------------------------------------
                   SIMULATION FAILED - ABORTED              
    --------------------------------------------------------

MikeOpenHWGroup, 2020-10-26:
My apologies @Silabs-ArjanB, the documentation in core-v-verif/cv32/sim/uvmt_cv32 on running tests is ridiculously out of date.

Pull-request #306 partially resolves this issue and tomorrow I'll update it to fully resolve the issue.  Once @strichmo has reviewed it, I'll merge it onto the master and this problem will resolve itself.

MikeOpenHWGroup, 2020-11-02:
All Compliance tests are in the cv32_static_compliance_regression and passing in regression.

closed 2020-11-02 by MikeOpenHWGroup