openhwfoundation/core-v-verif #304: minstret_count property

MikeOpenHWGroup, 2020-10-24:
As reported in https://github.com/openhwgroup/core-v-verif/pull/303, a recent update to hpmcounter_basic_test.c  triggers the a_minstret_count  assertion in `uvmt_cv32e40p_debug_assert.sv`.   Please investigate and either report a bug agains the RTL or fix the property.

### Steps to Reproduce
```
git clone https://github.com/openhwgroup/core-v-verif/
cd core-v-verif/cv32/sim/uvmt_cv32
make test TEST=hpmcounter_basic_test
```

Be sure to get the latest core-v-verif as the update to hpmcounter_basic_test was just recently merged in.

silabs-oysteink, 2020-10-25:
This property is already fixed in my branch that will be PR'ed together with Steve's fixes soon. If you would like me to fix this property earlier to get coverage running, please let me know and I'll make a separate branch for the properties.

MikeOpenHWGroup, 2020-10-25:
Thanks for the update @silabs-oysteink.  No need to create a separate PR, let's wait for the one you've already planned.

silabs-oysteink, 2020-10-27:
This should now be fixed in master branch.

MikeOpenHWGroup, 2020-10-27:
Indeed.  Test passes on hash 3f5b8e4 of core-v-verif.   Good work all.  Closing.

closed 2020-10-27 by MikeOpenHWGroup