openhwfoundation/core-v-verif #319: Counters test-program fails in the presense of memory stalls

MikeOpenHWGroup, 2020-10-30:
I am not yet certain this sighting is a real issue, but I want to record this information for later analysis.

The `counters` testcase (cv32/tests/programs/custom/counters) has been passing reliably for a couple of weeks now.  The test is now failing intermittently in regression when stalls on the memory interface are enabled.  
```
UVM_INFO @ 1.000 ns : mm_ram.sv(211) reporter [RNDSTALL] INSTR OBI stall enable: 1
UVM_INFO @ 1.000 ns : mm_ram.sv(212) reporter [RNDSTALL] INSTR OBI stall mode:   1
UVM_INFO @ 1.000 ns : mm_ram.sv(213) reporter [RNDSTALL] INSTR OBI stall gnt:    1
UVM_INFO @ 1.000 ns : mm_ram.sv(214) reporter [RNDSTALL] INSTR OBI stall valid:  0
UVM_INFO @ 1.000 ns : mm_ram.sv(215) reporter [RNDSTALL] INSTR OBI stall max:    2
UVM_INFO @ 1.000 ns : mm_ram.sv(216) reporter [RNDSTALL] DATA  OBI stall enable: 1
UVM_INFO @ 1.000 ns : mm_ram.sv(217) reporter [RNDSTALL] DATA  OBI stall mode:   2
UVM_INFO @ 1.000 ns : mm_ram.sv(218) reporter [RNDSTALL] DATA  OBI stall gnt:    0
UVM_INFO @ 1.000 ns : mm_ram.sv(219) reporter [RNDSTALL] DATA  OBI stall valid:  1
UVM_INFO @ 1.000 ns : mm_ram.sv(220) reporter [RNDSTALL] DATA  OBI stall max:    1
```

The failure signature is:
```
instret count = 4
pass
Jump register hazards count = 0
fail
```

### Steps to Reproduce
Fails on the head of the master branch of core-v-verif
```
make test SIMULATOR=dsim TEST=counters DSIM_SEED=982826100
```

### Additional context
This  test and seed combination always fails in the same manner with **dsim 20200720.11.0** under multiple platforms.  It is a given that the same seed will not produce the same results with different simulators.

MikeOpenHWGroup, 2020-10-31:
Update: in a subsequent regression with dsim (seed = 605238458), the random stalls were on the Instruction bus only and the failure signature was the same.  This does not prove anything but it indicates that the issue is sensitive to instruction fetches, not data read/writes.
```
UVM_INFO @ 1.000 ns : mm_ram.sv(211) reporter [RNDSTALL] INSTR OBI stall enable: 1
UVM_INFO @ 1.000 ns : mm_ram.sv(212) reporter [RNDSTALL] INSTR OBI stall mode:   1
UVM_INFO @ 1.000 ns : mm_ram.sv(213) reporter [RNDSTALL] INSTR OBI stall gnt:    1
UVM_INFO @ 1.000 ns : mm_ram.sv(214) reporter [RNDSTALL] INSTR OBI stall valid:  0
UVM_INFO @ 1.000 ns : mm_ram.sv(215) reporter [RNDSTALL] INSTR OBI stall max:    1
UVM_INFO @ 1.000 ns : mm_ram.sv(216) reporter [RNDSTALL] DATA  OBI stall enable: 0
UVM_INFO @ 1.000 ns : mm_ram.sv(217) reporter [RNDSTALL] DATA  OBI stall mode:   0
UVM_INFO @ 1.000 ns : mm_ram.sv(218) reporter [RNDSTALL] DATA  OBI stall gnt:    0
UVM_INFO @ 1.000 ns : mm_ram.sv(219) reporter [RNDSTALL] DATA  OBI stall valid:  0
UVM_INFO @ 1.000 ns : mm_ram.sv(220) reporter [RNDSTALL] DATA  OBI stall max:    0
```

Silabs-ArjanB, 2020-11-02:
Hi @MikeOpenHWGroup I can have a look at this if you want. I expect to confirm that this is expected behavior (as there cannot be a jump register hazard if the jump itself arrives in the pipeline late due to instruction fetch stalls. I would also at first glance expect that the load-use hazard could fail in a similar manner if stalls are present, but maybe this works a bit differently. If confirmed maybe we can break the test in two tests (one that is tolerant against wait states and one part that is not)

MikeOpenHWGroup, 2020-11-02:
That would be appreciated @Silabs-ArjanB, thanks!

The path to the test program is `core-v-verif/cv32/tests/programs/custom/counters/counters.c` and the original author is @eroom1966.

Silabs-ArjanB, 2020-11-04:
Hi @MikeOpenHWGroup I see that the test is basically an out-of-date copy of the one in core-v-verif/cv32/tests/programs/custom/hpmcounter_basic_test, so one of these two needs to be removed. Even if it had been up to date the test would fail under the presence of memory stalls and I can confirm that the jump hazard test itself is wrong. My proposal will be to fix hpmcounter_basic_test so that it will pass both with and without wait states (by relaxing the assembly test a bit), and then add an additional test that only checks if the jump hazard count is correct in the scenario that there are not wait states. We could modify counters.c to be that very limited test. 

I don't know how/where to state that such new test needs to be run without instruction wait states. Can you please advice on that?

I still need to investigate why the load-use test does _not_ fail (as I expected it would fail in a similar manner).

PilotAware, 2020-11-04:
I think this test was a copy (from another test) which I created, I used this to prove the mechanism of reading hmpcounter values from the RTL, as the RM does not have a cycle accurate version of these.
I am guessing this test is probably superceded from the full test ?

eroom1966, 2020-11-04:
I think this test was a copy (from another test) which I created, I used this to prove the mechanism of reading hmpcounter values from the RTL, as the RM does not have a cycle accurate version of these.
I am guessing this test is probably superceded from the full test ?

Silabs-ArjanB, 2020-11-04:
It is an out of date copy from code I originally wrote myself (that is why I am debugging it)

Silabs-ArjanB, 2020-11-05:
I can confirm that the test was also failing for 'load use' hazards (as expected).

Once https://github.com/openhwgroup/core-v-verif/pull/333 has been finished this issue can be closed.

closed 2020-11-06 by MikeOpenHWGroup