openhwfoundation/core-v-verif #331: Step and compare fails if debug_req_i is asserted at the same time as a trigger match

silabs-oysteink, 2020-11-05:
## Bug Title
Step and compare fails if debug_req_i is asserted at the same time as a trigger match

### Component
**Component:Verif**

### Steps to Reproduce
1. Clone https://github.com/silabs-oysteink/core-v-verif
2. Check out the branch silabs-oysteink_debug-cov-update
3. make  test TEST=debug_test USE_ISS=YES

Note that my fork of core-v-verif currently points to my fork of the cv32e40p repo as there is a (not yet PR'ed) change to the tracer to support ebreak instructions at the trigger address.

>   test7.3: Expect Trigger and debug_req_i
UVM_ERROR @ 461124.300 ns : uvmt_cv32_step_compare.sv(96) reporter [Step-and-Compare] dcsr expected=0x400080c3 and actual=0x40008083 PC=0x1a110800

The miscompare translates to RTL using cause=2 (trigger) while the ISS is using cause=3 (haltreq). In a race between these to causes, trigger should win since it has the highest priority.
[image]

MikeOpenHWGroup, 2020-11-05:
Good catch @silabs-oysteink.  Is the problem in the Reference Model, the tracer, or the step-and-compare logic?   It seems from your last sentence, the issue is with the RM not recognizing that trigger has a higher priority.  However, it could be that the RM doesn't see coincident trigger and halt events.

silabs-oysteink, 2020-11-05:
I think that the issue is within the ISS, Lee is on the case to check it out. There is a secondary issue though, as communicated per e-mail; The code generated by different versions of GCC may break the exact timing for the debug_req and irq from the virtual peripheral.

MikeOpenHWGroup, 2020-11-17:
Hi @silabs-oysteink, @eroom1966, can one or both of you post an update to this issue?   It is important that we resolve this issue this week.

silabs-oysteink, 2020-11-17:
The test "debug_test_trigger" is now passing with ISS enabled, so to my knowledge this issue is fixed. Maybe @eroom1966 can comment on the solution?

eroom1966, 2020-11-17:
yes this apears to be passing

MikeOpenHWGroup, 2020-11-17:
Thanks @silabs-oysteink and @eroom1966.  I will close this issue now.

closed 2020-11-17 by MikeOpenHWGroup