openhwfoundation/core-v-verif #337: Improvements: Test Architecture

strichmo, 2020-11-05:
### Use this issue to gather all ideas to improve or highlight long-term issues with any of the following:
- Testbench methodology
- UVM Agents
- UVM testbench
- UVM envirionments

**Add comments for all ideas to this issue.  Please try to refrain from deeper discussion of inidividual items.  A separate issue should be opened to enable and record that discussion.**

strichmo, 2020-12-01:
Move all of the generated test code (e.g. .S file for corev-dv, .elf, .objdump, etc.) into the "results" area.

MikeOpenHWGroup, 2020-12-01:
> Move all of the generated test code (e.g. .S file for corev-dv, .elf, .objdump, etc.) into the "results" area.

Not sure I get that.  The current home is consistent with the manually generated code which creates a single mental model for the organization of test-programs.

MikeOpenHWGroup, 2020-12-01:
Organize test-programs hierarchically.  For example:
- tests/programs/custom/plup
- tests/programs/custom/debug
- tests/programs/corev-dv/interrupt