openhwfoundation/core-v-verif #372: Zero bin removed from imm operand of c.slli and s.srli covergroups

MikeOpenHWGroup, 2020-11-15:
In #370, the zero bin was removed from c_slli_cg and c_srli_cg.   The code snipts below illustrate:

Before...
```
    covergroup c_srli_cg     with function sample(ins_t ins);
        option.per_instance = 1;
        cp_rd    : coverpoint get_gpr_name(ins.ops[0].val, ins.ops[0].key, "c.srli") {
            bins gprval[] = {[s0:a5]};
        }
        cp_rs1   : coverpoint get_gpr_name(ins.ops[1].val, ins.ops[1].key, "c.srli") {
            bins gprval[] = {[s0:a5]};
        }
        cp_shamt5   : coverpoint get_imm(ins.ops[2].val, "c.srli" ) {
            bins zero = {0};
            bins pos  = {[1:$]};
        }
    endgroup
```
After...
```
    covergroup c_srli_cg     with function sample(ins_t ins);
        option.per_instance = 1;
        cp_rd    : coverpoint get_gpr_name(ins.ops[0].val, ins.ops[0].key, "c.srli") {
            bins gprval[] = {[s0:a5]};
        }
        cp_rs1   : coverpoint get_gpr_name(ins.ops[1].val, ins.ops[1].key, "c.srli") {
            bins gprval[] = {[s0:a5]};
        }
        cp_shamt5   : coverpoint get_imm(ins.ops[2].val, "c.srli" ) {
            // Note that zero is reserved (used for HINTs)
            bins pos  = {[1:$]};
        }
    endgroup
```

It is not clear to me that we should not cover `c.srli rd uimm` where rd != x0 and uimm ==0.

strichmo, 2020-11-16:
Thanks for this.  You caught me...I meant to restore this based on your point about HINTs.  As Hint generation is now enabled in a couple of riscv-dv tests, I should restore these bins and ensure coverage.

strichmo, 2020-11-16:
Addressed here: https://github.com/openhwgroup/core-v-verif/pull/374

closed 2020-11-16 by strichmo

MikeOpenHWGroup, 2020-11-16:
Merged in.    Yeah, we've talked about this before - the GitHub process for reviewing check-ins really can improve quality.   At the expense of schedule I'm afraid...