openhwfoundation/core-v-verif #394: Downloaded Code from The Git repository and facing some compilation error

afzalj95-alt, 2020-11-21:
Hi All,

I have downloaded code from the git repository and following the command that has been suggested but I am facing this compilation error.

ERROR :- **# vsim -sv_seed 1 -suppress 7031 "+USE_ISS" -sv_lib /home/psingh86/core-v-verif-master/vendor_lib/imperas/riscv_CV32E40P_OVPsim/bin/Linux64/riscv_CV32E40P.dpi -sv_lib /cadtools/mentor/Questa_10.6b/questasim/verilog_src/uvm-1.2/src/../../../uvm-1.2/linux_x86_64/uvm_dpi -batch -do "/home/psingh86/core-v-verif-master/cv32/sim/tools/vsim/vsim.tcl" "+firmware=/home/psingh86/core-v-verif-master/cv32/tests/programs/custom/hello-world/hello-world.hex" "+elf_file=/home/psingh86/core-v-verif-master/cv32/tests/programs/custom/hello-world/hello-world.elf" -l vsim-hello-world.log "+UVM_TESTNAME=uvmt_cv32_firmware_test_c" uvmt_cv32_tb_vopt 
# Start time: 06:08:57 on Nov 21,2020
# ** Note: (vsim-3812) Design is being optimized...
# ** Error: (vopt-13130) Failed to find design unit uvmt_cv32_tb_vopt.**

[image]

If it helps below is the snapshot of the work directory that make file has generated.

[image]

Thanks

MikeOpenHWGroup, 2020-11-21:
Hi @afzalj95-alt, thanks for the issue.  I do not have access to Questa (vsim), so I am not able to reproduce this issue.  I have assigned the issue to @strichmo, who does have vsim access.

In the meantime, it would be helpful if you could provide the exact command-line you used.  For example, something like this:
```
$ git clone https://github.com/openhwgroup/core-v-verif master
$ cd master/cv32/sim/uvmt_cv32
$ make test TEST=hello-world SIMULATOR=vsim
```
Please also consult the [Simulation Environment README](https://github.com/openhwgroup/core-v-verif/tree/master/cv32/sim/uvmt_cv32#simulation-directory-for-cv32e-uvm-verification-environment) to ensure you have all the required dependencies.   For example, it is possible that you do not have the version of vsim that supports the full SystemVerilog LRM.   Our verification environment is written in SV/UVM so this is a requirement.    Also, the UVM environment uses an ISS from Imperas which is a licensed product.  You can disable use of the ISS at the command-line: `make test TEST=hello-world SIMULATOR=vsim USE_ISS=NO`.   Note that this will allow tests to run, but you will not be able to fully verify the core or collect functional coverage without it.

afzalj95-alt, 2020-11-23:
Hi Mike,

Thanks for the quick response.

I use the **git clone --recursive https://github.com/openhwgroup/core-v-verif** command to download code on my Linux terminal.
To run the simulation I have used  **make SIMULATOR=vsim hello-world** command from master/cv32/sim/uvmt_cv32 directory.

Also, I have tried to make test TEST=hello-world SIMULATOR=vsim USE_ISS=NO but with no success.

Thanks

MikeOpenHWGroup, 2020-11-23:
Your command lines look good to me @afzalj95-alt.  Note that the `--recursive` argument is not required as we do not use submodules in this repo.

@strichmo, are you able to reproduce this issue?

afzalj95-alt, 2020-11-24:
Hi @strichmo, Any update?

MikeOpenHWGroup, 2020-11-24:
Hi @afzalj95-alt, I cannot directly help, but I noticed that the "run" target in vsim.mk is dependent on RTLSRC_VOPT_TB_TOP which is not defined.  I suggest starting there.   If you are successful debugging the issue, please issue a pull-request and we can get the fix "in".

strichmo, 2020-11-24:
@afzalj95-alt I have no issues running with this command line combination:

% make clean_all
% make test TEST=hello-world SIMULATOR=vsim

Note that my version of Questa is quite a recent one: 2020.3_2

Please let me know which version of Questa you are using.

MikeOpenHWGroup, 2020-11-25:
Hi @afzalj95-alt.   Another member of the community had a similar issue, which is now resolved (see #397).   I noticed that person is also using an older version of Questa, so I expect you will be able to as well.   Keep in mind that you will also require an ISS license.  Please see the resolution of #397 for more information.

If we had not heard back from you by 2020-12-01, I will close this issue.   Good luck and please let us know your progress and if we can help further.

MikeOpenHWGroup, 2020-12-02:
> If we had not heard back from you by 2020-12-01, I will close this issue.

As we have not yet heard anything further on this issue, I am closing it.

closed 2020-12-02 by MikeOpenHWGroup