openhwfoundation/core-v-verif #397: Issue while passing UVM Test

prachi789, 2020-11-23:
Hi All,

I am facing issue while running UVM Test.
test :- hello-world
PATH : core-v-verif / cv32 / sim /  uvmt_cv32

Command used for run uvm test is :  make SIMULATOR=vsim hello-world

[image]

MikeOpenHWGroup, 2020-11-23:
Hi @prachi789, thanks for bringing this issue to our attention.   Your command-line is not exactly correct.  It should be:
```
make test SIMULATOR=vsim TEST=hello-world
```
Having said that, there is another person reporting a similar issue (#394) and we are looking into it.  Please follow that issue and I expect your issue will be resolved in the same manner.

prachi789, 2020-11-23:
Hi ,

I have tried the above command (make test SIMULATOR=vsim TEST=hello-world )
but it shows me the same error message.

Thanks & Regards,
Prachi  Talati

prachi789, 2020-11-24:
> Hi @prachi789, thanks for bringing this issue to our attention. Your command-line is not exactly correct. It should be:
>
> ```
> make test SIMULATOR=vsim TEST=hello-world
> ```
>
> Having said that, there is [...]

prachi789, 2020-11-24:
Hi Mike,

I've tried above command but it did not work. It shows me the same error.

[image]
[image]

strichmo, 2020-11-24:
Hi @prachi789 It appears you are using a very old Questa version: 10.6b.  Could you please try with a new Questa version if at all possible?

prachi789, 2020-11-25:
Hi strichmo,
Thanks for a response 
When I  have pass this command ( make test SIMULATOR=VSIM TEST=hello-world USE_ISS=NO) at that moment I am not getting any kind of error.

When I have pass this Command (make test SIMULATOR=VSIM TEST=hello-world ) at that moment I am facing license issue.

Thanks in advance.

[image]

MikeOpenHWGroup, 2020-11-25:
@prachi789, this is as expected.  As explained in the [README](https://github.com/openhwgroup/core-v-verif/blob/master/cv32/sim/uvmt_cv32/README.md) in the `cv32/sim/uvmt_cv32` directory, you will require an OVPsim license from Imperas to use the reference model (ISS).  There are instructions in the README (and in the error message you provided) for getting this license.

I see that things are now working as intended, so I close this issue.  Please have a look at the README and either contact me or Imperas directly.   (My email address is part of my GitHub profile and my GitHub user-name is provided in the README). Thanks!

closed 2020-11-25 by MikeOpenHWGroup