openhwfoundation/core-v-verif #405: Getting error while running assembly test

Prarthi1703, 2020-11-26:
I am facing error while running UVM test in assembly language (pupl_vectorial_add_sub)

test :- pupl_vectorial_add_sub
PATH : core-v-verif / cv32 / sim / uvmt_cv32

Command used for run uvm test is : make SIMULATOR=vsim pupl_vectorial_add_sub
[image]

MikeOpenHWGroup, 2020-11-26:
Hi @Prarthi1703.

One issue I can see for sure is that the format of the "make" command you are using is not correct.  The basic format to run a test is `make test SIMULATOR=<sim> TEST=<test-program>`.

Please review the [README](https://github.com/openhwgroup/core-v-verif/tree/master/cv32/sim/uvmt_cv32#simulation-directory-for-cv32e-uvm-verification-environment) for more information.

The error messages you are seeing are generated by the assembler, so it is probable that you do not have the proper toolchain installed.  If you want to run any of the PULP testcases, you must used the PULP toolchain.  The link provided above provides additional information you'll need to install the PULP toolchain on your computer.

Please be aware that the PULP extensions are still experimental.  As detailed in the [CV32E40P User Manual](https://core-v-docs-verif-strat.readthedocs.io/projects/cv32e40p_um/en/latest/integration.html), the PULP instructions have not yet been fully verified.  A future release of the CV32E4 core will support some or all of the PULP instructions.  In parallel with that, there is a project at OpenHW to integrate these PULP instructions into the mainstream RISC-V assembler toolchain.  If you are interested in the details about this, you can have a look at the [CORE-V software project](https://github.com/openhwgroup/core-v-sw).

Prarthi1703, 2020-11-27:
Hi Mike,

I have tried running test using  make test SIMULATOR=<sim> TEST=<test-program> command, still getting same error. 

I have installed latest version of riscv gnu toolchain from the link below.
https://github.com/riscv/riscv-gnu-toolchain.

MikeOpenHWGroup, 2020-11-27:
Hi @Prarthi1703, please read the README carefully.  If you are going to run test-programs that use instructions from the PULP ISA, you must use the PULP toochain.  The GNU toolchain will not compile PULP instructions.

MikeOpenHWGroup, 2020-12-01:
Hi @Prarthi1703, are you still experiencing difficulty?  If I have not heard back from you by the end of this week I will assume this issue can be closed.   Please let me know.

MikeOpenHWGroup, 2020-12-04:
Closing to due lack of activity.

closed 2020-12-04 by MikeOpenHWGroup