openhwfoundation/core-v-verif #407: ci_check anomoly

eroom1966, 2020-11-27:
I get a failure message if I do the following 

makecv32 clean_all
makecv32 sanity USE_ISS=YES
makecv32 test TEST=debug_test_trigger USE_ISS=YES
ci_check --keep --simulator xrun | tee ~/ci_check.log
CI Check FAILED: Expected 14 tests to run but found only 15 PASSED or FAILED messages

running the test manually in the middle messes up the couns, running this works

makecv32 clean_all
ci_check --keep --simulator xrun | tee ~/ci_check.log

Thx
Lee

MikeOpenHWGroup, 2020-11-27:
Hi @eroom1966, I am 99% confident that the issue is related to your use of the `--keep` switch.  If you had any logfiles from previous runs, these could confuse the script as it assumes that any logs it finds in cv32/sim/uvmt_cv32/<sim>_results are part of the regression.  My guess is that you had a logfile from a previous run which is why "only 15" (out of 14!) passed.

The purest in me dislikes the "keep" switch as anything run from a pre-compiled version of the DUT cannot be deemed a proper regression.  I'll push in a quick update to the script to warn users that use of that.

eroom1966, 2020-11-27:
Aha, I thought the keep was only related to the rtl
Thanks and closing

closed 2020-11-27 by eroom1966