openhwfoundation/core-v-verif #409: Please add "compliance" target to vcs.mk

MikeOpenHWGroup, 2020-11-30:
Hi @masgia.  As of #406 the compliance regression has the ability to dump compliance test signatures and compare them to the reference signatures provided in compliance test-suite.   More information is available in the [UVM README](https://github.com/openhwgroup/core-v-verif/tree/master/cv32/sim/uvmt_cv32#risc-v-compliance-test-suite-and-regressions).

This update requires a simulator-specific `compliance` target to be added to each \<sim\>.mk   Can you add this for vcs.mk?   Check out xrun, dsim or vsim for a working example.

Thanks!

masgia, 2020-11-30:
Hi,

OK, I'll add the feature

MikeOpenHWGroup, 2020-12-01:
Thanks @masgia, I've merged in #415 so we are good to go.

closed 2020-12-01 by MikeOpenHWGroup