openhwfoundation/core-v-verif #410: Please add "compliance" target to riviera.mk

MikeOpenHWGroup, 2020-11-30:
Hi @dawidzim .  As of #406 the compliance regression has the ability to dump compliance test signatures and compare them to the reference signatures provided in compliance test-suite.   More information is available in the [UVM README](https://github.com/openhwgroup/core-v-verif/tree/master/cv32/sim/uvmt_cv32#risc-v-compliance-test-suite-and-regressions).

This update requires a simulator-specific `compliance` target to be added to each \<sim\>.mk   Can you add this for riviera.mk?   Check out xrun, dsim or vsim for a working example.

Thanks!

dawidzim, 2020-12-01:
Hey @MikeOpenHWGroup . Sure, I will do this

MikeOpenHWGroup, 2020-12-03:
Thanks @dawidzim.  #419 is merged in.

closed 2020-12-03 by MikeOpenHWGroup