openhwfoundation/core-v-verif #412: Ignored tests in compliance

Imperas, 2020-12-01:
@MikeOpenHWGroup 

Action to be completed at some point

There are currently 4 tests in the compliance suite which are ignored for cv32e40p.
This is due to an inadequacy in the tests and the test infrastructure, NOT cv32e40p

I think there are 2 issues
1. The tests expect to write any address value to mtvec, we could fix this by having an alignment definition per target
2. The tests expect to trap on the instruction behavior in the test

For misaligned load store, cv32e40p will resolve this in hardware as part of the BIU, whereas the test expects the misalignment to raise a trap, so the behavior differs and therefore will fail

For misaligned fetches, the test attempts to execute an instruction at a half-word aligned address, and raise an exception
In the case of cv32e40p, support for the C extension, means that fetches from half-word aligned addresses is legal, and therefore no fetch misaligned trap is raised.
it may subsequently result in an illegal instruction exception, rather than a misaligned fetch, so again the behavior will differ.

For the ECALL and EBREAK, the test assumes that DEBUG mode is not supported, and therefore will cause a trap
In the cv32e40p DEBUG mode is supported

Silabs-ArjanB, 2020-12-01:
Hi, can a corresponding issue (r issues) be opend at the RISC-V compliance repos, so that this eventually gets fixed?

MikeOpenHWGroup, 2020-12-01:
Its not clear to me that the RISC-V Compliance owners will see this as something they should fix:

- the way cv32e40p handles misaligned load/stores may be viewed as particular to the cv32e40p.
- not all cores will support the C extension so the misaligned test is valid for them.
- not all cores will support Debug Mode so the assumed behavior of ECALL and EBREAK is not "wrong".

A better use of our time is to understand the deltas between what the Compliance test-suite exercises and what our specific core needs verified and do that.   This we have done, so I leave it to others to engage the Compliance group on this particular topic.

Silabs-ArjanB, 2020-12-01:
Hi @MikeOpenHWGroup , for the misaligned load/store issue the test is clearly wrong and should get fixed (which has been debated for more than a year already and the relevant ticket got closed without a fix). The one related to the C extension is clear a test issue as well and should be fixed at their side. ECALL has nothing to do with debug mode really, so I will not accept above explanation as to why that test is failing. EBREAK has to do with debug, but again I do not think the reason for the test failure is properly described in this ticket (I believe it has to do with mtvec alignment restrictions in our core). If it indeed has to do with allowed mtvec alignment restrictions then it becomes a test case issue as well and it needs to be filed in the proper place.

I think it is really as simple as: Either the cv32e40p RTL is wrong or the test case is wrong (in the sense that it should not make assumptions about what other extensions might or might not be present); there is not too much room for a grey area here (other than that the BSP or custom code that we might run could be wrong). We cannot claim that the test cases are wrong without letting the RISC-V compliance owners know and give them a chance to react.

MikeOpenHWGroup, 2020-12-01:
> We cannot claim that the test cases are wrong without letting the RISC-V compliance owners know and give them a chance to react.

@eroom1966 _is_ a Compliance owner.  They are well aware.

Imperas, 2020-12-01:
@Silabs-ArjanB @MikeOpenHWGroup  As in my initial posting
>  This is due to an inadequacy in the tests and the test infrastructure, NOT cv32e40p

Let me drive this through the compliance working group, they need to fix this.

Silabs-ArjanB, 2020-12-02:
Hi @Imperas Thank you Lee. As far as I am concerned this waiver is okay then (I cannot close it myself as I do not have such permissions in this repos, nor do I think I should have such permissions). Anyway, it is okay to close this one and I am looking forward to the response from the Compliance Working Group.

Imperas, 2020-12-02:
Hi All
I have posted the following https://github.com/riscv/riscv-compliance/issues/149 which references the 4 tests individually.
I have asked that these tests be deprecated or improved to cover the spec correctly.
I think to cover the spec correctly will be quite a lot of work, and so deprecation seems the best plan
Thx
Lee

MikeOpenHWGroup, 2020-12-02:
Thanks guys.   I'll close this issue now - just a reminder that doing so implies an acceptance of the waiver.

closed 2020-12-02 by MikeOpenHWGroup