openhwfoundation/core-v-verif #425: FSM coverage waiver for cv32e40p

strichmo, 2020-12-07:
The following report contains the FSM waivers for code coverage for the cv32e40p. These waivers are a combination of generated waivers using formal unreachability analysis and waivers generated from manual code inspection.

In the repository these waivers are in these files:
cv32/sim/tools/xrun/cv32e40p_no_pulp.auto.vRefine
cv32/sim/tools/xrun/cv32e40p_no_pulp.manual.vRefine

https://mikeopenhwgroup.github.io/core-v-docs/Waivers/cv32e40p_fsm_waiver/index.html

MikeOpenHWGroup, 2020-12-09:
Updated URL for the link to the waivers.

silabs-robin, 2023-12-11:
@MikeOpenHWGroup Can his be closed?

silabs-robin, 2023-12-11:
See also https://github.com/openhwgroup/core-v-verif/issues/424 and https://github.com/openhwgroup/core-v-verif/issues/423