openhwfoundation/core-v-verif #433: Use vcs-mx (version:O-2018.09-SP2) run :make hello-world, occurs "Illegal instruction (core 0) at PC 0x000001e8:"

JmheYouzi, 2020-12-10:
# Template for CORE-V-Verif Issues

## Bug Title
A clear and concise description of what the bug is.
Use vcs-mx (version:O-2018.09-SP2) run :make hello-world, occurs "Illegal instruction (core 0) at PC 0x000001e8:"
### Type
* Confusing or extraneous status or error messages

### Steps to Reproduce
Please provide:
1. URL to branch that exhibits the issue.
 https://github.com/openhwgroup/core-v-verif/tree/master/cv32/sim/uvmt_cv32

2. Command line:  
 make hello-world

3. Logfile and/or wave-dump info (screen shots can be useful)
[image]
[image]
[image]

### Additional context
    My CentOS version is 6.1, and when I type : make hello-world, it occurs the line " Illegal instruction (core 0) at PC 0x000001e8‘’ ，I want to know how to fix this problem.Thanks for your effort!

MikeOpenHWGroup, 2020-12-10:
Hi @JmheYouzi, thanks for your interest in core-v-verif.   I suspect your problem is a combination of two things:
1. Your command-line is incorrect.  Please review the `README` in the `uvmt_cv32` directory, especially the [test-programs](https://github.com/openhwgroup/core-v-verif/tree/master/cv32/sim/uvmt_cv32#available-test-programs) section.
2. You must use an approved [Toolchain](https://github.com/openhwgroup/core-v-verif/blob/master/cv32/sim/TOOLCHAIN.md) to compile/link your code.

Let me know if these resolve your issues.

JmheYouzi, 2020-12-10:
Thank  you very much,I fixed this problem , and  I can move on to leran the UVM environment.
First,this time I use the correct command-line：make test TEST=hello-world
Second, I dowmload the  approved Toolchain and set env in "/etc/profile"
[image]
[image]
[image]

closed 2020-12-10 by JmheYouzi