openhwfoundation/core-v-verif #434: "UVM_ERROR @ 777.300 ns : mm_ram.sv(315) reporter [MM_RAM] Cannot open signature file C-JAL.signature_output for writing (error_str: Too many open files). "

JmheYouzi, 2020-12-11:
## Bug Title
When i run :make compliance_regression RISCV_ISA=rv32imc. occurs two same UVM_ERROR

"UVM_ERROR @ 777.300 ns : mm_ram.sv(315) reporter [MM_RAM] Cannot open signature file C-JAL.signature_output for writing (error_str: Too many open files).
"

### Type
Indicate whether the type of problem you found:
* Confusing or extraneous status or error messages

### Steps to Reproduce
Please provide:
1. URL to branch that exhibits the issue:
https://github.com/openhwgroup/core-v-verif/tree/master/cv32/sim/uvmt_cv32

2. Command line:
make compliance_regression RISCV_ISA=rv32imc

3. Logfile and/or wave-dump info (screen shots can be useful)
[image]
[image]
[image]
[image]
[image]

### Additional context
Add any other context about the problem here.
First , I can success run "make test TEST=hello-world" ,and PASSED, but when I run "make compliance_regression RISCV_ISA=rv32imc" , it FAILED,  and occurs such UVM_ERROR, I want to know how to fix this problem.Looking forward to your reply,thanks !

MikeOpenHWGroup, 2020-12-11:
This is an OS issue.  Our environment opens a _lot_ of files as it executes and the compliance tests opens additional files to dump and compare signature files as well. does not always close them properly.  If you have others tools that also do this you may run into a limit imposed by your OS.   I am not sure which OS you are using, but you may find [this article](https://askubuntu.com/questions/181215/too-many-open-files-how-to-find-the-culprit) helpful.

Question: are you using the ISS?

JmheYouzi, 2020-12-12:
Thanks , my OS version is CentOS 6.10. I haven't know am I used the ISS.

MikeOpenHWGroup, 2020-12-12:
Please let me know if increasing the number of allowed file descriptors resolves this issue.   If I have not heard from you within 7 days I will assume the problem is resolved and close this issue.

JmheYouzi, 2020-12-13:
Thank you, I have fixed this problem.

closed 2020-12-13 by JmheYouzi