openhwfoundation/core-v-verif #441: Install CORE-V Toolchain in CentOS7

JackShanZemin, 2021-01-04:
Hello 

I want to run core-v-verif in CentOS7，and I have already installed vcs and verdi in CentOS7. But I found that 

the instructure of install RISC-V GCC Compilers is based on ubuntu，can I have a instructure based on CentOS7？

Best Regards 
Jack

MikeOpenHWGroup, 2021-01-04:
The [README](https://github.com/openhwgroup/core-v-verif/blob/master/cv32/sim/uvmt_cv32/README.md) in the UVM simulation directory has a section about the [GCC tools required](https://github.com/openhwgroup/core-v-verif/tree/master/cv32/sim/uvmt_cv32#risc-v-gcc-compilers), and there is a link to a set of [detailed instructions](https://github.com/openhwgroup/core-v-verif/blob/master/cv32/sim/TOOLCHAIN.md) for obtaining the required toolchain.  If you follow those, you can find a link to a set pre-built toolchains for the CORE-V cross-compiler (toolchain), including a [CentOS7 version](https://www.embecosm.com/resources/tool-chain-downloads/#corev).

You can also build your own from source, but I do not think this will be necessary.

JackShanZemin, 2021-01-04:
hello Mike

I can find the CentOS7 version riscv tool chain，but I do not know how to install it in CentOS7, can you give a structure?

Best Regards
Jack

MikeOpenHWGroup, 2021-01-04:
Its a pre-compiled binary, so its simple:
- find a place to put it (I recommend /opt/corev, but it can be anywhere you like)
- create a shell environment variable COREV to point to it.

JackShanZemin, 2021-01-05:
hello mike

I do not know ，how to create a shell enviroment variable COREV to point it？can you tell me this ？

Best Regards
Jack

MikeOpenHWGroup, 2021-01-05:
It depends on the shell environment you are using.  For bash use `export COREV=/opt/corev`.  You will need at least some familiarity with Linux and scripting tools such as python3, Make, etc. to be successful with core-v-verif.  There are many on-line resources for this.

JackShanZemin, 2021-01-07:
hello mike，

I generate testbench by "make test SIMULATOR=vcs TEST=hello-world".

and I got this error " /opt/riscv/bin/riscv32-unknown-elf-gcc：command not found ".

Best Regards
Jack

MikeOpenHWGroup, 2021-01-07:
Do have a toolchain installed at /opt/riscv?  If so, do you have an environment variable pointing to it?

Please review the "Toolchain" comment header in `core-v-verif/cv32/sim/Common.mk`.  If you want to use the recommended toolchain then you should have a toolchain installed at `/opt/corev` and the environment variable **COREV** should be set to "YES".   With this in place, the Makefile will compile your test-program with `/opt/corev/bin/riscv32-core-v-elf-gcc`.

JackShanZemin, 2021-01-07:
hello mike

I installed tool chain in at /opt/riscv， but I am not sure wheatehr it is ture,

first I copy all the toll chain folder such as bin,include and riscv32-corev-elf into /opt/riscv,

second I add comment :

PATH="/opt/riscv/bin:$PATH"
export RISCV="/opt/riscv"

in .bashrc file, and source .bashrc.

is it right?

I use centos7

Best Rehards
Jack

MikeOpenHWGroup, 2021-01-07:
No, that is not right.  You are trying to mix the setup for a RISCV toolchain with a COREV toolchain.   You can use either (COREV is recommended), but the structure of these toolchains are different.

The RISCV toolchain:
- By convention is installed at /opt/riscv
- Is pointed to by $RISCV (i.e. RISCV=/opt/riscv)
- The machine architecture is set to "unknown" (therefore /opt/riscv/bin/riscv32-unknown-elf-gcc is used to compile your program)

The COREV toolchain:
- By convention is installed at /opt/corev
- Will be used if $COREV is set to YES (i.e. export COREV=YES)
- The machine architecture is set to "corev" (therefore /opt/corev/bin/riscv32-corev-elf-gcc is used to compile your program)

I have run this successfully on CentOS7, Debian9, Debian10, Ubuntu18.04 and Ubuntu20.04, so you should be able to run on your OS.

jeremybennett, 2021-01-07:
Hi @JackShanZemin 

Depending on how you did the copying, links may or may not be set up properly. Where are you getting your tool chain? If it is one of the [Embecosm pre-compiled binaries](https://www.embecosm.com/resources/tool-chain-downloads/), then just unpack it directly into `/opt/riscv`.

In your `.bashrc` you only need to set up PATH to use the compiler tool chain. @MikeOpenHWGroup  can advise if you also need to set up the `RISCV` environment variable, but it is not needed for the compiler.

I don't think you have to install in `/opt/riscv`, you can put the tool chain anywhere, so long as you set the path.

HTH,

Jeremy

closed 2021-01-07 by jeremybennett

JackShanZemin, 2021-01-09:
hello mike

As I  configuraed all the enviroment, I try to generate test case, I use 

command "make test SIMULATOR=vcs TEST=hello-world", then I got folder 

vcs_results.

I think UVM simulate work is finished ,as the attachment picture show.

But how can I start verdi wave, I try to use command "make verdi" in directer

/core-v-verif-master/cv32/sim/,but I did not start verdi. 

And how can I get coverage repor?

Best regards
Jack

MikeOpenHWGroup, 2021-01-09:
Glad to see you are making progress.

For information about support for debug and coverage tools, please refer to the [build-configurations](https://github.com/openhwgroup/core-v-verif/tree/master/cv32/sim/uvmt_cv32#build-configurations) section of the README in  the `/sim/uvm` directory.

If you have additional questions about this, please reach out to me via email (available on my GitHub profile).  These issues are not intended for support conversations.  :wink:

JackShanZemin, 2021-01-10:
hello mike

First, I use command line "make test SIMULATOR=vcs TEST=hello-world WAVE=1 GUI=1 COV=1 ",

then I got debug window as picture show, if it possible that I can get wave like verdi?

I use ADV_DEBUG=YES flag to start verdi window, but I GOT error info "verdi is not support vcs".

Second, I try to get coverage report, I type "make cov TEST=hello-world GUI=1" as picture show,

is it right? and where can I get the coverage report? in directory /sim/uvmt_cv32/vcs_result/?

Best Regards
Jack

MikeOpenHWGroup, 2021-01-10:
Hi Jack, as I said previously, these issues are not the place for such discussions.  Please each out to me via email (available on my GitHub profile).