openhwfoundation/core-v-verif #443: What is data value in objdump file

Prarthi1703, 2021-01-11:
Hi,

I am running test for odd even number code in core-v-verif and understanding test flow

In objdump file I am enable to understand data value in instruction set
[image]

MikeOpenHWGroup, 2021-01-11:
Hi @Prarthi1703, thanks for your question.   The objdump is generated by the Toolchain.  So it is a tool used by OpenHW for core-v-verif, but not developed by OpenHW.  There is lots of good information about objdump online, including [this Wikipedia article](https://en.wikipedia.org/wiki/Objdump).

closed 2021-01-11 by MikeOpenHWGroup

Prarthi1703, 2021-01-12:
Hi Mike,

Thanks for responding.

My doubt is what is 6555 value meaning in objdump file
[image]

MikeOpenHWGroup, 2021-01-12:
Hi @Prarthi1703, I _could_ tell you, but I am not a consultant for objdump users (and not a compiler toolchain expert in any case).   If you want to understand the output format of objdump (or any other component of the compiler toolchain), please consult the relevant documentation.