openhwfoundation/core-v-verif #447: Problem with debug tests

dawidzim, 2021-01-26:
Hey!
When I call command:
`make SIMULATOR=riviera USE_ISS=0 test TEST=debug_test` this error show up:
`make: *** No rule to make target '/home/dawidz/github/core-v-verif/cv32/tests/programs/custom/debug_test/debug_test_0.hex', needed by 'test'.  Stop.`
If I call `debug_test_reset` work fine.
Also in file `cv32\regress\cv32_debug.yaml` is wrong header:
```
# Header
name: cv32_interrupt
description: Directed and random interrupt tests for CV32E40P
```

MikeOpenHWGroup, 2021-01-26:
Thanks for the bug report @dawidzim.   #448 _should_ resolve your issue.   You can clone by fix to see for yourself, or wait for the merge.

dawidzim, 2021-01-27:
hey @MikeOpenHWGroup 
#448 works for me :)
thanks!

closed 2021-01-27 by dawidzim