openhwfoundation/core-v-verif #452: "Running the testbench with Verilator" isn't working

scfmoe, 2021-02-01:
# "Running the testbench with Verilator" isn't working
I followed all readmes provided on this git and tried to run some basic simulations of the cv32e40p with verilator.
It stops with fatal errors in file mm_ram.sv.

## make custom CUSTOM_PROG=hello-world didn't run
no Verilator command runs as expected: It always stops with fatal error in line 449 and 456 of mm_ram.sv
"core-v-verif/cv32/tb/core/mm_ram.sv:449: $fatal only allowed under an assertion."
additionally in line 336. Can't find /open file  "sig_file"

### Type
Indicate whether the type of problem you found:
* Confusing or extraneous status or error messages

### Steps to Reproduce
1. URL to branch that exhibits the issue: ...\core-v-verif\cv32\sim\core
2. Command line:  make / make custom CUSTOM_PROG=hello-world
3.
[image]

### Additional context
I downloaded Verilator via "apt-get install verilator" on a "Ubuntu 16.04 LTS" Subsystem under windows 10.

My goal is to run custom programs with help of verilator on a modified cv32e40p, to verifiy the core is still working/ the modifed core works as intended.
But the example programs provided on this git aren't working for me with the unmodified cv32e40p.
I couldn't find a hint inside all readmes and followed all instructions to set the simulation in \core-v-verif\cv32\sim\core up.

jeremybennett, 2021-02-01:
@scfmoe I'm not sure which version of Verilator ships with Ubuntu 16.04, but it might be quite old (I believe it ships with Verilator 3.874-1 from June 2015).

The CORE-V RTL does use some constructs only supported in more recent versions of Verilator. I'm currently using 4.035. Can you try using a newer version of Verilator and see if you still have the same problems. It is not a difficult tool to build from source. Instructions here: https://www.veripool.org/projects/verilator/wiki/Installing.

MikeOpenHWGroup, 2021-02-01:
Hi @scfmoe, thanks for the bug report.  I am very interested to see a member of the community working to get core-v-verif running under a Windows environment.   AFAIK, this has not been attempted before and so I am happy to help.  Once you are successful it would be great to document your experience in the appropriate README to help others.

As @jeremybennett reports, you will need an up-to-date version of Verilator.  I am currently running `Verilator 4.032 2020-04-04 rev v4.032` under Ubuntu 20.04 and have also run the same version of Verilator under Ubuntu 18.04, Debian 9 and Debian 10.

I have found it best to build both Verilator and gtkwaves from source as the pre-built versions are typically out of date.

> My goal is to run custom programs with help of verilator on a modified cv32e40p, to verifiy the core is still working/ the modifed core works as intended.

Very cool.  This is one of the intended goals of core-v-verif.  Please be aware that the "core" testbench does not have the ability to run our full suite of tests and is not intended to fully verify the core.  The "core" testbench is a demonstration vehicle only.  If you intend to full verify your modified core, you will need to use the UVM environment and Verilator currently does not support that.

scfmoe, 2021-02-01:
Hey @MikeOpenHWGroup, Hey @jeremybennett,
currently I'm installing Ubuntu 20.04 Subsystem on Windows and try to re-intstall Verilator and try the makefile again.

I need the simulation for my bachelor thesis. The goal of my work is to write a Built-in Self Test for a RISC-V Core.
To check the BIST module I looked at verilator, because modelsim and the Vivado Simualtor are not working and I can't use expensive software (e.g. questasim).
The Built-in self Test module should be able to test the ALU. All other parts remain untouched.

When I finished my bachelor thesis I could try to write a readme and share my experiences, but for now I have more imortant work to do.

MikeOpenHWGroup, 2021-02-02:
Hi @scfmoe, your project sounds very interesting.  Please let us know how it goes.  You can reach me at mike@openhwgroup.org.

Please let me know if I can close this issue.  If you do not update this issue in the next week, I will close it unilaterally.

scfmoe, 2021-02-03:
Hey, 
current state is: I installed a new subsystem with Ubuntu 20.04 and Verilator 4.032 2020-04-04 rev v4.032. 
The former problem is solved. 
The test is not running yet, but I think I have to re-install the correct riscv-toolchain.
So I guess we can close this Issue.
Thanks for your help! I let you know how it is going via e-mail.

closed 2021-02-03 by scfmoe

jeremybennett, 2021-02-03:
@scfmoe Good to hear. Pre-compiled tool chains are available on the Embecosm website: https://www.embecosm.com/resources/tool-chain-downloads/

scfmoe, 2021-02-05:
Hey,

Thanks for the pre-compiled toolchain. All tests proposed in the ReadMe are now running without any problem.
Thank you for your help!